pjalocha/esp32-ogn-tracker #1: Build failed

ubx, 2018-03-05:
I tried to build/configure the software after the description in [http://wiki.glidernet.org/esp32-ogn-tracker](url)

But "make menuconfig" failed:
> Makefile:8: /make/project.mk: No such file or directory
> make: *** No rule to make target '/make/project.mk'.  Stop.
> 

Thanks
Andreas

pjalocha, 2018-03-05:
This looks like there is no path to esp-idf defined ?

ubx, 2018-03-05:
No, it wasn't. Sorry for asking stupid questions. Now it works!

closed 2018-03-05 by ubx

AviraMatti, 2019-06-17:
Hi Andreas,

I have about same kind problem 'make menuconfig' donn't work.
Message is  " *** No rule to make target  'menuconfig'  Stop  "

I add  /.profile file two line more...as next additional lines...what has in Diy tracker page..
All have done to follow introduction in Diy Tracker page.
I have last Obuntu laptop.

export PATH=$PATH:~/xtensa-esp32-elf-raspberry/xtensa-esp32-elf/bin
export IDF_PATH=~/esp-idf
export PATH=$PATH:~/xtensa-esp32-elf/xtensa-esp32-elf/bin
export IDF_PATH=~/esp-idf/make

Please, have you help for me?.

Matti
my direct email matttkos@gmail.com

AviraMatti, 2019-06-17:
.....in my file those lines are not overwrote...as previous message...

Matti