pjalocha/esp32-ogn-tracker #3: Building issue

slm0001, 2019-03-09:
Hi,

I try compile your code on Debian and get the following error message:
*************************************************************************************
/root/esp/esp32-ogn-tracker/main/hal.cpp: In function 'int BT_SPP_Init()':
/root/esp/esp32-ogn-tracker/main/hal.cpp:506:56: error: 'struct esp_bt_controller_config_t' has no member named 'mode'
   Err = esp_bt_controller_enable((esp_bt_mode_t)BTconf.mode); if(Err!=ESP_OK) return Err;   // mode must be same as in BTconf
                                                        ^
/root/esp/esp32-ogn-tracker/main/hal.cpp:514:3: error: 'esp_bt_sp_param_t' was not declared in this scope
   esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
   ^
/root/esp/esp32-ogn-tracker/main/hal.cpp:515:3: error: 'esp_bt_io_cap_t' was not declared in this scope
   esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_NONE; // _IO;
   ^
/root/esp/esp32-ogn-tracker/main/hal.cpp:516:33: error: 'param_type' was not declared in this scope
   esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
                                 ^
/root/esp/esp32-ogn-tracker/main/hal.cpp:516:46: error: 'iocap' was not declared in this scope
   esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
                                              ^
/root/esp/esp32-ogn-tracker/main/hal.cpp:516:68: error: 'esp_bt_gap_set_security_param' was not declared in this scope
   esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
*********************************************************************************************************

What did I wrong? Can you help me?

Thanks in advance,
SzL

pjalocha, 2019-03-09:
I have updated the ESP-IDF and the compiled as it is in the repository right now, and it did compile.
Are you sure you use the most recent ESP-IDF ?
If not: type there: git pull and git submodule update

slm0001, 2019-03-10:
I use the latest stable version of ESP-IDF. Should I use the development branch? :-)

pjalocha, 2019-03-10:
Well, this could be the problem, I actually use the most recent thus development branch
thus I do git pull from this address:
https://github.com/espressif/esp-idf.git

slm0001, 2019-05-07:
Hi Pavel,

I updated my Debian, so I have the latest versions, so finally I could build the code.
I can see the NMEA messages and also the OLED shows detailed info.
My question is: How can I send "$POGNS,Pilot=Pawel,FreqCorr=-20,Reg=EC-AC9,Model=JANUS" command to the tracker? Via linux terminal or any other method?
Thanks in advance,
SzL

pjalocha, 2019-05-07:
I just type it in :-)
or type it on the side and the copy-paste into the terminal window.
thus simplest is just copy what you have typed above and paste it into the terminal
then press Ctrl-C to confirm visually the changes.

the important point, when you setup minicom, is to disable the hardware and software handshake
otherwise what you type is not sent as the hardware handshaking lines are not active.

Let me know if it works.

slm0001, 2019-05-07:
On Linux I could not manage it, but on Windows via Termite application I could! :-)

helijunky, 2019-05-07:
You can type `make monitor` to start IDF Monitor and enter commands in the same way as minicom.
[https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html)

sb00000, 2022-08-18:
Hello, I get error when running make command: /esp32-ogn-tracker/main/disp.cpp:29:28: error: 'DEFAULT_DispPage' was not declared in this scope
static uint8 t DISP_Page = DEFAULT_DispPage ; Fab501 before 0