pjalocha/esp32-ogn-tracker #4: Compile error WITH_MAVLINK

helijunky, 2019-05-02:
Hi Pawel

when compiling WITH_MAVLINK I get the following error:

`esp/esp32-ogn-tracker/main/gps.cpp:591:51: error: 'class GPS_Position' has no member named 'getUnixTime_ms'
     { uint64_t PrevUnixTime_ms = Position[PosIdx].getUnixTime_ms();`

any ideas?
thx
Manuel

pjalocha, 2019-05-02:
I think I fixed it just now, but I did not test it, if you can, please do so, and let me know of any troubles.
I did not compile the MAVlink for some time, this is why this thing came up.

Pawel.

helijunky, 2019-05-02:
Thanks for the quick fix, it compiles now. I will test it as soon as possible.
However, there's another issue: since the major update of January 22, 2019, my tracker (TTGO) is not visible to ground stations anymore. GPS works fine and it looks like it sends out OGN packets, but there must be something wrong with it (either packet format or transmitter disabled?). Previous version from July 23, 2018 works fine.
Thanks for your support,
Manuel

pjalocha, 2019-05-02:
This can be frequency correction, I find the TTGO and HELTEC modules need some frequency correction, most often of 20kHz - did you need to do this correction with the given board.
And, now, I changed the units for this correction in the parameters, thus please check what is the FreqCorr written or not written there. Do you have local OGN receiver, even with tiny antenna,, would be helpful to check this.

pjalocha, 2019-05-02:
You could try to set in blind -20ppm frequency correction by sending the following to the USB serial console:
$POGNS,FreqCorr=-20
and then confirm byCtrl-C to see the parameters.
-20ppm is the most often frequency correction I see with those modules.

helijunky, 2019-05-02:
Bingo! The old value of -17kHz was translated into -170ppm after the update. I changed it back to -17ppm and it works perfectly. And yes, I have my own local OGN receiver. I can now see the tracker in the aircraft-list with a deviation of around +0.1kHz which is not too bad, is it?

pjalocha, 2019-05-02:
Great !

0.1kHz is not bad, is like 0.1ppm off.
OGN receivers have tolerance in the order of +/-15-20kHz thus this is not a problem at all.
I will surely change with temperature but should not more than few ppm which is still fine.

Let us know how it works otherwise. I checked it with older Ardupilot which for example would send incorrect time and no Geoid-Separation, and the baro sensor was "cheap" and produced pressures smaller by a factor of two, so I have to correct for all this: you can see this in the parameters.

Pawel.

helijunky, 2019-05-05:
I've tested today with Arduplane 3.9.8 on a Pixhawk Flightcontroller. The HELTEC was connected to Telemetry Port 2 of the Pixhawk. The display shows GPS position and time and barometric altitude, all coming from the Ardupilot. Also the device is visible on the OGN receiver. So far so good.
However, ADSB_VEHICLE messages from other nearby trackers are no longer coming. I found that the corresponding code is commented out since the major release of January:
https://github.com/pjalocha/esp32-ogn-tracker/blame/dabfffb230bb8def4b64638f458d887bf65590eb/main/proc.cpp#L283
Uncommenting doesn't solve the issue. Could you please have a look?

thx
Manuel

pjalocha, 2019-05-05:
Can you give any more details: is there a message sent at all or no message at all ?

pjalocha, 2019-05-05:
It should get better now, the MAVlink message was sent to the console, and not the MAVlink port, thus at least the message should be seen now. Possibly this is not all which needs to be done but this first step is there already.

helijunky, 2019-05-06:
still no ADSB_VEHICLE message coming through. Here's the output from monitor, 8DDEC0 being the ID of the nearby tracker:

`$POGNR,1,5,,-92.5,108,+31,,*6D`
`$POGNR,1,5,,-93.0,110,+31,,*60`
`$POGNT,16,1,3,8DDEC0,0,11,1.5,4704.0924N,00717.8568E,446,,+0.1,0.1,000.0,+0.0,-38,0*6E`
`$PFLAA,0,-2,+14,+6,3,8DDEC0,000.0,+0.0,0.1,+0.1,1*66`
`$POGNR,1,7,,-93.0,112,+31,,*60`
`$POGNR,1,7,,-92.5,114,+31,,*62`
`$POGNT,18,1,3,8DDEC0,0,11,1.5,4704.0924N,00717.8568E,446,,-0.2,0.1,000.0,+0.0,-37,0*6A`
`$PFLAA,0,-2,+14,+6,3,8DDEC0,000.0,+0.0,0.1,-0.2,1*63`
`$POGNR,1,8,,-92.5,116,+31,,*6F`
`$POGNR,1,8,,-93.0,118,+31,,*65`

pjalocha, 2019-05-09:
Do you see anything on the MAVlink port ?
Can you monitor/dump what is coming there from the Tracker ?

helijunky, 2019-05-09:
Should be fixed now. ADSB_VEHICLE messages are sent to the flight controller and further to the ground control station. See Pull Request #5

closed 2019-10-21 by helijunky