pjalocha/esp32-ogn-tracker #6: PlatformIO compile error

mbatini, 2019-10-11:
Hi Pavel,
is it my configuration or some versione problem?

I am using latest platformio and espidf framework. Getting this:

src/main/hal.cpp: In function 'void esp_spp_cb(esp_spp_cb_event_t, esp_spp_cb_param_t*)':
src/main/hal.cpp:404:32: error: 'ESP_BT_CONNECTABLE' was not declared in this scope
       esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
tia
Mauro

pjalocha, 2019-10-11:
It is the ESP-IDF version, at some point they expanded the call, thus either update ESP-IDF or put the old call, the two calls are:

      esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE); // for older ESP-IDF
      esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE); // for newer ESP-IDF

mbatini, 2019-10-11:
Thanks for the promptly answer.

Going to compile now

mbatini, 2019-10-11:
Ok, compile is ok after ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE was in place.

I've updated espressif32 platform and it compiles still ok

but it does not link:

Linking .pio/build/ttgo-lora32-v1/firmware.elf
.pio/build/ttgo-lora32-v1/src/main/ctrl.o:(.literal._Z10PrintTasksPFvcE+0xc): undefined reference to `uxTaskGetSystemState'
.pio/build/ttgo-lora32-v1/src/main/ctrl.o: In function `PrintTasks(void (*)(char))':
ctrl.cpp:(.text._Z10PrintTasksPFvcE+0x3f): undefined reference to `uxTaskGetSystemState'
.pio/build/ttgo-lora32-v1/src/main/hal.o:(.literal._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x14): undefined reference to `esp_bt_gap_set_scan_mode'
.pio/build/ttgo-lora32-v1/src/main/hal.o:(.literal._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x18): undefined reference to `esp_spp_start_srv'
.pio/build/ttgo-lora32-v1/src/main/hal.o:(.literal._Z15CONS_UART_Writec+0x4): undefined reference to `esp_spp_write'
.pio/build/ttgo-lora32-v1/src/main/hal.o:(.literal._Z11BT_SPP_Initv+0x14): undefined reference to `esp_bt_gap_register_callback'
.pio/build/ttgo-lora32-v1/src/main/hal.o:(.literal._Z11BT_SPP_Initv+0x18): undefined reference to `esp_spp_register_callback'
.pio/build/ttgo-lora32-v1/src/main/hal.o:(.literal._Z11BT_SPP_Initv+0x1c): undefined reference to `esp_spp_init'
.pio/build/ttgo-lora32-v1/src/main/hal.o:(.literal._Z11BT_SPP_Initv+0x20): undefined reference to `esp_bt_gap_set_security_param'
.pio/build/ttgo-lora32-v1/src/main/hal.o:(.literal._Z11BT_SPP_Initv+0x28): undefined reference to `esp_bt_gap_set_pin'
.pio/build/ttgo-lora32-v1/src/main/hal.o:(.literal._Z11BT_SPP_Initv+0x2c): undefined reference to `esp_bt_gap_set_cod'
.pio/build/ttgo-lora32-v1/src/main/hal.o: In function `esp_spp_cb(esp_spp_cb_event_t, esp_spp_cb_param_t*)':
hal.cpp:(.text._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x32): undefined reference to `esp_bt_gap_set_scan_mode'
hal.cpp:(.text._ZL10esp_spp_cb18esp_spp_cb_event_tP18esp_spp_cb_param_t+0x43): undefined reference to `esp_spp_start_srv'
.pio/build/ttgo-lora32-v1/src/main/hal.o: In function `CONS_UART_Write(char)':
hal.cpp:(.text._Z15CONS_UART_Writec+0xdc): undefined reference to `esp_spp_write'
.pio/build/ttgo-lora32-v1/src/main/hal.o: In function `BT_SPP_Init()':
hal.cpp:(.text._Z11BT_SPP_Initv+0x4c): undefined reference to `esp_bt_gap_register_callback'
hal.cpp:(.text._Z11BT_SPP_Initv+0x5b): undefined reference to `esp_spp_register_callback'
hal.cpp:(.text._Z11BT_SPP_Initv+0x66): undefined reference to `esp_spp_init'
hal.cpp:(.text._Z11BT_SPP_Initv+0x7b): undefined reference to `esp_bt_gap_set_security_param'
hal.cpp:(.text._Z11BT_SPP_Initv+0xaa): undefined reference to `esp_bt_gap_set_pin'
hal.cpp:(.text._Z11BT_SPP_Initv+0xbb): undefined reference to `esp_bt_gap_set_cod'
collect2: error: ld returned 1 exit status
*** [.pio/build/ttgo-lora32-v1/firmware.elf] Error 1

looks like ctrl.cpp and hal.cpp have linking problems. Do you have any hint?

pjalocha, 2019-10-11:
Possibly BT options in the sdkconfig thus via make menuconfig

Component config -> Bluetooth (enable)

Bluetooth Host (Bluedroid - Dual mode)

Bluedroid options -> Classic Bluetooth -> SPP