pjalocha/esp32-ogn-tracker #7: Make error

lupus78, 2019-12-17:
Hi, I follow the instructions and I think I have set up the toolchain correctly on my Ubuntu 18.04.

Where I run make in the code folder, I get the below error. Am I doing something wrong?

> lupus@alpina:~/esp/esp32-ogn-tracker$ make
> Toolchain path: /home/lupus/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
> WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a
> Expected to see version: esp-2019r2
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> WARNING: Compiler version is not supported: 5.2.0
> Expected to see version(s): 8.2.0
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> Python requirements from /home/lupus/esp/esp-idf/requirements.txt are satisfied.
> /home/lupus/esp/esp-idf/make/project.mk:640: warning: undefined variable 'IDF_SKIP_CHECK_SUBMODULES'
> WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a
> Expected to see version: esp-2019r2
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> WARNING: Compiler version is not supported: 5.2.0
> Expected to see version(s): 8.2.0
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> 
> App "app-template" version: 88b0918-dirty
> CXX build/main/hal.o
> /home/lupus/esp/esp32-ogn-tracker/main/hal.cpp: In function 'uint64_t getUniqueID()':
> /home/lupus/esp/esp32-ogn-tracker/main/hal.cpp:309:28: warning: unused variable 'ret' [-Wunused-variable]
>  { uint64_t ID=0; esp_err_t ret=esp_efuse_mac_get_default((uint8_t *)&ID); return ID; }
>                             ^
> /home/lupus/esp/esp32-ogn-tracker/main/hal.cpp: In function 'void RFM_TransferBlock(uint8_t*, uint8_t)':
> /home/lupus/esp/esp32-ogn-tracker/main/hal.cpp:646:13: warning: unused variable 'ret' [-Wunused-variable]
>    esp_err_t ret = spi_device_polling_transmit(RFM_SPI, &Trans); }
>              ^
> /home/lupus/esp/esp32-ogn-tracker/main/hal.cpp: In function 'int ADC_Init()':
> /home/lupus/esp/esp32-ogn-tracker/main/hal.cpp:1185:23: warning: unused variable 'val_type' [-Wunused-variable]
>    esp_adc_cal_value_t val_type = esp_adc_cal_characterize(ADC_unit, ADC_atten, ADC_WIDTH_BIT_12, ADC_Vref, ADC_characs); // calibrate ADC1
>                        ^
> /home/lupus/esp/esp32-ogn-tracker/main/hal.cpp: In function 'void IO_Configuration()':
> /home/lupus/esp/esp32-ogn-tracker/main/hal.cpp:1271:3: sorry, unimplemented: non-trivial designated initializers not supported
>    };
>    ^
> /home/lupus/esp/esp32-ogn-tracker/main/hal.cpp:1305:3: sorry, unimplemented: non-trivial designated initializers not supported
>    } ;
>    ^
> /home/lupus/esp/esp32-ogn-tracker/main/hal.cpp:1305:3: sorry, unimplemented: non-trivial designated initializers not supported
> /home/lupus/esp/esp32-ogn-tracker/main/hal.cpp:1305:3: warning: missing initializer for member 'i2c_config_t::<anonymous>' [-Wmissing-field-initializers]
> /home/lupus/esp/esp32-ogn-tracker/main/hal.cpp:1238:13: warning: variable 'ret' set but not used [-Wunused-but-set-variable]
>    esp_err_t ret=spi_bus_initialize(RFM_SPI_HOST, &BusCfg, 1);
>              ^
> At global scope:
> cc1plus: warning: unrecognized command line option '-Wno-frame-address'
> /home/lupus/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'hal.o' failed
> make[1]: *** [hal.o] Error 1
> /home/lupus/esp/esp-idf/make/project.mk:604: recipe for target 'component-main-build' failed
> make: *** [component-main-build] Error 2

lupus78, 2019-12-19:
In the meantime I tried it on Raspberry PI4 with latest Raspbian. Similar error message.

Also tried the Vagrant box, that fails before, when trying to install the Requirements for esp-idf.

pjalocha, 2019-12-19:
I tried to compile, with a little earlier ESP-IDF from November 2019 and it did compile.
With the most recent I have some issues thus I don't use it yet.
Try to remove the build directory and compile from scratch.
I can as well supply compiled binary for your platform

lupus78, 2019-12-19:
Deleting the "build" folder did not helped.

I want to modify your code, so I really want to be able to compile myself.
I guess I could try to get that "november" version of esp-idf, but I'm not sure if that would help, as there are many submodules, so I guess if I go back to an earlier commit on the main branch, the submodules will be checked out with the latest version. 
Just to give it a try, what is the last commit's id in your "november 2019" esp-idf?

pjalocha, 2019-12-19:
Most recent is commit 93a8603c545fb8e54741d6685146e2f3b874378d
I compile it now with x64 Ubuntu.

lupus78, 2019-12-19:
Wow, this worked. Thanks. 

Would be nice to make it work with the latest. It is frustrating to fail building it while one follows the wiki properly.

closed 2020-07-04 by lupus78