pjalocha/esp32-ogn-tracker #9: Make error :(

Hidvegi82, 2020-03-12:
Hi!
What i do wrong?
Ubuntu 18.04.4 LTS, x64
Thanks

ogn@ogn:~/esp32-ogn-tracker$ make
Toolchain path: /home/ogn/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2019r2
Compiler version: 8.2.0
Python requirements from /home/ogn/esp/esp-idf/requirements.txt are satisfied.
GENCONFIG
App "app-template" version: 86a24a2-dirty
CC build/bootloader/bootloader_support/src/bootloader_efuse_esp32.o

.........................................................................

CXX build/main/intmath.o
CXX build/main/gps.o
/home/ogn/esp32-ogn-tracker/main/gps.cpp:89:13: warning: 'GPS_Cmd' defined but not used [-Wunused-variable]
 static char GPS_Cmd[64];
             ^~~~~~~
CXX build/main/bitcount.o
CXX build/main/ldpc.o
CXX build/main/proc.o
/home/ogn/esp32-ogn-tracker/main/proc.cpp:295:16: warning: 'uint8_t WritePFLAU(char*, uint8_t)' defined but not used [-Wunused-function]
 static uint8_t WritePFLAU(char *NMEA, uint8_t GPS=1)    // produce the (mostly dummy) PFLAU to satisfy XCsoar and LK8000
                ^~~~~~~~~~
CXX build/main/aero.o
CXX build/main/rf.o
CXX build/main/st7789.o
/home/ogn/esp32-ogn-tracker/main/st7789.cpp:188:1: warning: missing initializer for member 'ledc_timer_config_t::clk_cfg' [-Wmissing-field-initializers]
 } ;
 ^
/home/ogn/esp32-ogn-tracker/main/st7789.cpp: In function 'void lcd_trans_wait()':
/home/ogn/esp32-ogn-tracker/main/st7789.cpp:282:15: warning: unused variable 'ret' [-Wunused-variable]
   { esp_err_t ret=spi_device_get_trans_result(LCD_SPI, &rtrans, portMAX_DELAY); }
               ^~~
/home/ogn/esp32-ogn-tracker/main/st7789.cpp: In function 'void lcd_trans_start()':
/home/ogn/esp32-ogn-tracker/main/st7789.cpp:288:15: warning: unused variable 'ret' [-Wunused-variable]
   { esp_err_t ret=spi_device_queue_trans(LCD_SPI, &lcd_trans[x], portMAX_DELAY); }
               ^~~
/home/ogn/esp32-ogn-tracker/main/st7789.cpp: In function 'UINT tjd_output(JDEC*, void*, JRECT*)':
/home/ogn/esp32-ogn-tracker/main/st7789.cpp:603:7: warning: unused variable 'Jys' [-Wunused-variable]
   int Jys = Ly1-Ly0;                     // Y-size of the JPEG rectagle
       ^~~
/home/ogn/esp32-ogn-tracker/main/st7789.cpp:607:7: warning: variable 'CropRight' set but not used [-Wunused-but-set-variable]
   int CropRight  = 0; if(Lx1>LCD_WIDTH)  { CropRight = LCD_WIDTH-Lx1; Lx1=LCD_WIDTH; }    //
       ^~~~~~~~~
/home/ogn/esp32-ogn-tracker/main/st7789.cpp:608:7: warning: variable 'CropBottom' set but not used [-Wunused-but-set-variable]
   int CropBottom = 0; if(Ly1>LCD_HEIGHT) { CropBottom = LCD_HEIGHT-Ly1; Ly1=LCD_HEIGHT; } //
       ^~~~~~~~~~
/home/ogn/esp32-ogn-tracker/main/st7789.cpp: In function 'void LCD_DrawJPEG(const uint8_t*, int, int, int, int)':
/home/ogn/esp32-ogn-tracker/main/st7789.cpp:648:11: warning: unused variable 'rc' [-Wunused-variable]
   JRESULT rc = jd_decomp(&decoder, tjd_output, Scale);
           ^~
/home/ogn/esp32-ogn-tracker/main/st7789.cpp: In function 'void LCD_Init(spi_host_device_t, uint8_t, int)':
/home/ogn/esp32-ogn-tracker/main/st7789.cpp:683:13: warning: unused variable 'ret' [-Wunused-variable]
   esp_err_t ret = spi_bus_add_device(LCD_SPI_HOST, &DevConfig, &LCD_SPI);     // Attach the LCD to the SPI bus
             ^~~
CXX build/main/format.o
/home/ogn/esp32-ogn-tracker/main/format.cpp:144:21: warning: integer constant is so large that it is unsigned
   for( Pos=20, Base=10000000000000000000; Base; Base/=10, Pos--)
                     ^~~~~~~~~~~~~~~~~~~~
CXX build/main/ognconv.o
CXX build/main/timesync.o
CXX build/main/disp_oled.o
/home/ogn/esp32-ogn-tracker/main/disp_oled.cpp:24:13: warning: 'Line' defined but not used [-Wunused-variable]
 static char Line[128];
             ^~~~
CXX build/main/atmosphere.o
CXX build/main/hal.o
/home/ogn/esp32-ogn-tracker/main/hal.cpp: In function 'uint64_t getUniqueID()':
/home/ogn/esp32-ogn-tracker/main/hal.cpp:483:28: warning: unused variable 'ret' [-Wunused-variable]
 { uint64_t ID=0; esp_err_t ret=esp_efuse_mac_get_default((uint8_t *)&ID); return ID; }
                            ^~~
/home/ogn/esp32-ogn-tracker/main/hal.cpp: In function 'void RFM_TransferBlock(uint8_t*, uint8_t)':
/home/ogn/esp32-ogn-tracker/main/hal.cpp:842:13: warning: unused variable 'ret' [-Wunused-variable]
   esp_err_t ret = spi_device_transmit(RFM_SPI, &Trans); }
             ^~~
/home/ogn/esp32-ogn-tracker/main/hal.cpp: At global scope:
/home/ogn/esp32-ogn-tracker/main/hal.cpp:952:3: warning: missing initializer for member 'ledc_timer_config_t::clk_cfg' [-Wmissing-field-initializers]
   } ;
   ^
/home/ogn/esp32-ogn-tracker/main/hal.cpp: In function 'int ADC_Init()':
/home/ogn/esp32-ogn-tracker/main/hal.cpp:1553:23: warning: unused variable 'val_type' [-Wunused-variable]
   esp_adc_cal_value_t val_type = esp_adc_cal_characterize(ADC_unit, ADC_atten, ADC_WIDTH_BIT_12, ADC_Vref, ADC_characs); // calibrate ADC1
                       ^~~~~~~~
/home/ogn/esp32-ogn-tracker/main/hal.cpp: In function 'void IO_Configuration()':
/home/ogn/esp32-ogn-tracker/main/hal.cpp:1670:3: warning: missing initializer for member 'i2c_config_t::scl_io_num' [-Wmissing-field-initializers]
   } ;
   ^
/home/ogn/esp32-ogn-tracker/main/hal.cpp:1670:3: warning: missing initializer for member 'i2c_config_t::scl_pullup_en' [-Wmissing-field-initializers]
/home/ogn/esp32-ogn-tracker/main/hal.cpp:1670:3: warning: missing initializer for member 'i2c_config_t::<anonymous>' [-Wmissing-field-initializers]
/home/ogn/esp32-ogn-tracker/main/hal.cpp:1670:3: error: designator order for field 'i2c_config_t::scl_io_num' does not match declaration order in 'i2c_config_t'
/home/ogn/esp32-ogn-tracker/main/hal.cpp:1695:11: warning: unused variable 'PwrStatus' [-Wunused-variable]
   uint8_t PwrStatus = AXP.readStatus();
           ^~~~~~~~~
/home/ogn/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'hal.o' failed
make[1]: *** [hal.o] Error 1
/home/ogn/esp/esp-idf/make/project.mk:635: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2

pjalocha, 2020-03-12:
Likely a different version of ESP-IDF, I tried the most recent and had to flip two lines around the line 1670 of hal.cpp it should be as below: the order of the values changed in the past and now it changed back.

#if defined(PIN_I2C_SCL) && defined(PIN_I2C_SDA)
  i2c_config_t I2C_Config =                            // I2C for OLED and pressue sensor
  { mode          : I2C_MODE_MASTER,
    sda_io_num    : PIN_I2C_SDA,
    scl_io_num    : PIN_I2C_SCL,
    sda_pullup_en : GPIO_PULLUP_ENABLE,
    scl_pullup_en : GPIO_PULLUP_ENABLE
  } ;
  I2C_Config.master.clk_speed =  I2C_SPEED;
  i2c_param_config  (I2C_BUS, &I2C_Config);
  i2c_driver_install(I2C_BUS, I2C_Config.mode, 0, 0, 0);
#endif

Hidvegi82, 2020-03-13:
This worked! Thank you very much.

closed 2020-03-13 by Hidvegi82