pjalocha/esp32-ogn-tracker #14: Battery

Hidvegi82, 2020-04-26:
Hi,

I use Heltec Lora 32 v2 with Oled. 
I tried it with LiPo and Li-ion batteries but the display show 0% and ~0.3V (+00.0mV/min).
The device runs on battery well and charges from usb.

Thanks!

pjalocha, 2020-04-26:
From what I know this module does not have the battery sense, thus resistor divider connected to input 35 or 36 like for example the older version of TTGO T-Beam. Or maybe it has ? It could be as well connected externally: thus from two 100k resistors from the battery terminal to GPIO #36.

Hidvegi82, 2020-04-26:
Thank you very much, it works!

Just one more question, sorry... Can I change the time zone? (It is here utc+2h)

pjalocha, 2020-04-26:
Did you connect the resistor divider to pin #36 and it shows the correct voltage ?
Time zone on the OLED ? No, it is always UTC time, thus "Z" at the end.

Hidvegi82, 2020-04-26:
Yes, i did connect 2 resistors from battery port to GPIO#36, and now, i can see the correct voltage and percent.
[image]

"Time zone on the OLED ? No, it is always UTC time, thus "Z" at the end."
Oh, so I can understand what the "Z" at the end. :D

pjalocha, 2020-04-26:
Great, thank you for the picture.
The percentage is based on the battery voltage and the relation is simplified: anything above 4.1V is considered 100%, anything below 3.6V is considered empty and in between the two the conversion is linear.

sueppchen, 2020-05-22:
you do not need to solder anything, the Battery-sense-pin changed to gpio37
also you will have to switch on the power for output (gpio21)
schematica found at 
[https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/SchematicDiagram/WiFi_LoRa_32(V2)/WIFI_LoRa_32_V2(868-915).PDF](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/SchematicDiagram/WiFi_LoRa_32(V2)/WIFI_LoRa_32_V2(868-915).PDF
)
changed a proc.cpp
in **_static void ReadStatus(OGN_Packet &Packet)_** :
```
#ifdef WITH_ESP32
    // Packet.clrTemperature();

    #ifdef WITH_HELTEC_V2
     uint16_t BattVolt = 1.6 * BatterySense();                                        // [mV] measure battery voltage
    #else
     uint16_t BattVolt = BatterySense();                                        // [mV] measure battery voltage
    #endif
```
changed hal.cpp (with some cleanup because it was bit confusing):

**_header_** is now sorted by boards : 
```

/*
https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

GPIO  HELTEC v2   HELTEC      TTGO       JACEK     M5_JACEK    T-Beam     T-Beamv10    FollowMe   Restrictions

 0    Button      Button      Button                .           LCD/DC     LCD/DC                  Can lock the boot, better not pull it down
 1    CONS/TxD    CONS/TxD    CONS/TxD   CONS/TxD   CONS/TxD    CONS/TxD   CONS/TxD                Console/Program
 2                                       SD/MISO    .           LCD/MOSI   LCD/MOSI     LED/DBG    Bootstrap: LOW to enter UART download mode
 3    CONS/RxD    CONS/RxD    CONS/RxD   CONS/RxD   CONS/RxD    CONS/RxD   CONS/RxD                Console/Program
 4    OLED/SDA    OLED/SDA    OLED/SDA   ADC/CS                 LCD/BCKL   LCD/BCKL     PERF/RST
 5    RF/SCK      RF/SCK      RF/SCK     RF/SCK     GPS/ANT     RF/SCK     RF/SCK       RF/CS                          PWM at boot
 6                                                  SD/CLK                              SD/CLK     SD/CLK              SPI flash
 7                                                  SD/DATA0                                       SD/DATA0            SPI flash
 8                                                  SD/DATA1                                       SD/DATA1            SPI flash
 9                                                  SD/DATA2                                       SD/DATA2            SPI flash
10                                                  SD/DATA3                                       SD/DATA3            SPI flash
11                                                  SD/CMD                              SD/CMD     SD/CMD              SPI flash
12    GPS/TxD     GPS/RxD     GPS/RxD    SD/CS      GPS/RxD                GPS/TxD      SD/MISO    HSPI/MISO JTAG/TDI  Bootstrap: select output voltage to power the flash chip
13                GPS/Ena     GPS/Ena    SD/SCK     LCD/SCL     LCD/CLK    LCD/CLK      SD/MOSI    HSPI/MOSI JTAG/TCK
14    RF/RST      RF/RST      RF/RST     Beeper     ?           LED/PCB    .            SD/CLK     HSPI/CLK  JTAG/TMS  PWM at boot
15    OLED/SCL    OLED/SCL    OLED/SCL   SD/MOSI    GPS/TxD                .            HSPI/CS0   JTAG/TDO            PWM at boot
16    OLED/RST    OLED/RST    OLED/RST   RF/IRQ                 GPS/Tx     RAM/CS       GPS/TX     U2_RXD
17    Beeper      Beeper      Beeper     RF/RST                 GPS/Rx     Beeper       GPS/RX     U2_TXD
18    RF/CS       RF/CS       RF/CS      RF/MISO    RF/CS       RF/SCK     RF/CS                   VSPI/CLK
19    RF/MISO     RF/MISO     RF/MISO    RF/MOSI    RF/MISO     RF/MISO    RF/MISO      RF/MISO    VSPI/MISO
20                                                                                                 not listed
21    Vext_p_on   .           LED?       RF/CS      I2C/SCL     I2C/SDA    I2C/SDA      I2C/SDA    VSPI/QUADHP
22    .           .           .          PWR/ON     I2C/SDA     I2C/CLK    I2C/SCL                 VSPI/QUADWP
23    .           .           .          PWR/LDO    RF/RST      RF/MOSI    RF/RST       RF/MOSI    VSPI/MOSI
24                                                                                                 not listed
25    LED         LED         Speaker    .          Speaker     Speaker    Speaker      TT/RX
26    RF/IRQ      RF/IRQ      RF/IRQ     SCL        RF/IRQ      RF/IRQ     RF/IRQ       PWR/Good
27    RF/MOSI     RF/MOSI     RF/MOSI    SDA                    RF/MOSI    RF/MOSI      TT/TX
28                                                                                                 not listed
29                                                                                                 not listed
30                                                                                                 not listed
31                                                                                                 not listed
32    .           .           .          GPS/TxD    ?                      .            TT/RST     XTAL
33    .           .           .          OLED/RST   ?                      .            GPS/EN     XTAL
34    RF/DIO2     GPS/PPS     GPS/PPS    GPS/RxD    KNOB/Sense             GPS/TxD      GPS/PPS
35    RF/DIO1     GPS/TxD     GPS/TxD    GPS/PPS    BAT/Sense              AXP/IRQ      RF/IRQ
36                .          BAT/Sense  BAT/Ext.               .            Vbat/Sense
37    BAT/Sense            .                                            GPS/PPS
[... cut ...]
39    GPS/RxD     .                     Button      Button     Button

*/

[... 309 lines cut ...]
#endif
```

in _**void IO_Configuration(void)**_
```
#ifdef WITH_JACEK
  POWER_LDO_Dir();                  // speific power control
  POWER_LDO_On();
  POWER_Dir();
  POWER_On();
#endif

#ifdef WITH_HELTEC_V2
  POWER_Dir();
  POWER_On();
#endif

#ifdef WITH_M5_JACEK
  GPS_ANT_Dir();
  GPS_ANT_Sel(1);                   // 0 = external entenna
#endif

```

everything is running fine but without outdoor-test

sueppchen, 2020-05-24:
sorry, I made an mistake in section //ADC
must be
 ```
#ifdef WITH_TBEAM
 static adc1_channel_t ADC_Chan_Batt = ADC1_GPIO35_CHANNEL;
 static adc1_channel_t ADC_Chan_Knob = ADC1_GPIO34_CHANNEL;
#else
 #ifdef WITH_HELTEC_V2
  static adc1_channel_t ADC_Chan_Batt  = ADC1_GPIO37_CHANNEL;
 #else
  static adc1_channel_t ADC_Chan_Batt  = ADC1_GPIO36_CHANNEL;
 #endif
#endif
```

Hidvegi82, 2020-05-24:
Hi
I tested the changes today, but the gpio36 still needed the resistor divider.
Then you sent the correction (in section ADC), and now it works perfectly.
Tomorrow I'm doing an outdoor test.
Thank you!

Hidvegi82, 2020-05-27:
Sorry, it rained for 2 days.
I did an outdoor test today.
The display shows battery data OK, BMP280 data OK, GPS data OK. But the tracker did not appear on the map even next to the reciever.
Although http://wiki.glidernet.org/esp32-ogn-tracker "With Heltec V2 Board the parameter FreqCorr by default is ok (0), do not change.", after i set FreqCorr to -20, appeared the tracker on the map.
So everything is perfect!
I have a little problem ... When the battery is fully charged, the charge indicator LED will continue to light. Is this controlled by software or the panel?
This is a 3.7V Li-Po battery, I will try it with an 18650 Li-ion battery.
Thanks again for your help!