pjalocha/esp32-ogn-tracker #16: Custom OLED page

lupus78, 2020-04-27:
Pawel, 

What you think about adding a "blank" page to the OLED screen for custom content?

For my flying (I'm a paraglider) I only need a few paramteres, and I'd like to see them quickly. So I've made an extra page with only: altitude, climb rate and speed. You can see my code here: https://github.com/lupus78/esp32-ogn-tracker/tree/pgscreen

I'm thinking to have this as a custom screen, so anyone could put where they want in there. 
What do you think?

pjalocha, 2020-04-27:
I think this is fine, post a picture of such page.
Ideally there could be many pages which are being enabled/disabled by for example longer or double press on the button and the configuration could be stored in the parameters.
We could for example say we have 32 pages and each can be enabled/disabled and the state can be stored as 32-bit word, each bit saying the page is enabled/disabled.

lupus78, 2020-04-27:
That's a great idea. Here is my screen:

[image]

pjalocha, 2020-04-27:
Add this page, I will add a parameter to configure which pages are to be visible when switching with the push button. Maybe push the altitude to the left and add the direction about the speed ?

lupus78, 2020-04-27:
Can we make it so that you can set the order of these screens? For example I would put this on the first page, everything else is nice to have but no need while in flight.

Direction of speed, you mean the heading?