pjalocha/esp32-ogn-tracker #18: Bluetooth Serial commands

lupus78, 2020-06-17:
Pawel, am I right that you can not send configuration commands via the Bluetooth Serial interface?

I tried but no effect. See the attached screenshot, $POGNS,Verbose=0 should turn off the messages.

[image]

Paulie92, 2020-07-03:
Hi,

 I heve exactly the same problem. I was also trying to send the command via Arduino IDE serial monitor also without success. But I am wondering, what should mean what is mentioned at pawel's instructions: _"To list all parameters with their values send Ctrl-C (**software and hardware handshake must be OFF**)."_ The problem could be here, but I don't know what the "handshake" is and how to disable it. Does anyone have an idea or hint?

lupus78, 2020-07-03:
That is not the same issue. Via the USB cable the terminal works. The Aurdino IDE is not suitable to send the Ctrl-C command, you should use the make's built in serial monitor or minicom to send the Ctrl-C and other key combinations. Read this for more details: http://wiki.glidernet.org/esp32-ogn-tracker

pjalocha, 2020-07-03:
I confirm there is no data path from the BT incoming data to the console input, thus I shall create it, possibly today.

Paulie92, 2020-07-03:
The new version does not work for me now at all. I even cannot cannot connect via bluetooth. Also via the the USB serial connection I cannot see the NMEA data flow. It looks lile the ESP is only rebooting again and again.

pjalocha, 2020-07-03:
Yes, sorry, a bug, fixed now, please try again.

Paulie92, 2020-07-04:
The ESP is running now but the bluetooth commands still not for me.

lupus78, 2020-07-04:
Pawel, thanks the Bluetooth it working! This opens up the possibility to make a desktop or mobile app for configuration. 

I suggest to move someof the functional settings from config.h to parameters. Like setting which NMEA sentence to send, or config for Vario parameters. What do you think?