pjalocha/esp32-ogn-tracker #22: What is PPS input used for?

smurry, 2020-07-15:
Hello, Pawel,

I am very new to your project, but managed to build a first prototype tracker quite easily using your instructions.  Thanks very much for this amazing work.

I am planning to build several additional units to be used for our glider club (which lacks FLARM in most of the ships used solely for training).  For these, I need to use a GPS receiver that lacks the PPS output.  I believe from reading your code that this will not be a problem, as I can simply comment out the  "#define WITH_GPS_PPS" line in the config.h file.

My question is, what functions will be lacking without the PPS enabled?  Will there be any reduction in capability without this PPS signal?

Thanks again for this excellent work. I only wish my C programming skills were sharp enough to help your development.

pjalocha, 2020-07-15:
Great !

PPS is not absolutely needed. it is for exact timing for transmitting packets on the two frequencies: 868.2 and 868.4MHz in Europe.
If you don't use PPS the timing will be taken from the data how it appears on the serial port of the GPS receiver.
There is a parameter you can adjust for this: PPSdelay, the default value is 100ms.
This is the time from the true PPS (or beginning of the UTC second) to the moment the first byte from the GPS appears on the UART.
Using this "soft" PPS yields worse timing accuracy but it is enough to transmit a signal recognized by the OGN network.
The reception by other OGN-Treackers can be impaired a little as they will switch to the other frequency at a bit different moment, but this is not a serious issue, occasionally some packet would not be received  by other devices.

smurry, 2020-07-16:
Thanks, Pawel.  I really appreciate the information.  In fact, I realized after writing to you yesterday that the tracker that I built already had mis-wired the PPS signal.  I am using a Heltec V2 and mistakenly wired the PPS to GPIO34 instead of GPIO38.  Nevertheless it seems to be working OK.  Perhaps not idea, but it is sending position updates that are being received correctly.  So, I guess that the PPS signal really isn't necessary as you said.  Anyway, I'm building a second version and will take note of your advice above.  Thanks for the help.

pjalocha, 2020-07-16:
When the GPS is not sending a PPS the OGN-Tracker extracts the timing from the timing of the UART data.
It is less precise but enough to make the system to work.
So when you misswired the PPS, then it still had the timing.