pjalocha/esp32-ogn-tracker #26: Satellite problem

Hidvegi82, 2020-10-02:
Hello. 

Maybe I did stupid things... 
I ordered a T-beam ttgo motherboard (T22_V11-2019.1212) with Meshtastic software on it.
I uploaded an esp-32-ogn software that I used before, but I didn't see any satellites with it. I reinstalled Meshtastic, seeing the satellites again.
Unfortunately, I installed this version of Meshtastic on a working OGN, it also saw the satellites with it, but after I put your esp32-ogn-tracker software back, it doesn't see the satellites anymore.
The gps signal led flashes.
Any idea why the satellites aren't visible? No hardware changes were made.

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5360
ho 0 tail 12 room 4
load:0x40078000,len:14240
load:0x40080400,len:3892
entry 0x40080684
[1B][0;32mI (378) cpu_start: Pro cpu up.[1B][0m
[1B][0;32mI (378) cpu_start: Application information:[1B][0m
[1B][0;32mI (378) cpu_start: Project name:     app-template[1B][0m
[1B][0;32mI (381) cpu_start: App version:      2cd55cf-dirty[1B][0m
[1B][0;32mI (386) cpu_start: Compile time:     Oct  1 2020 16:56:31[1B][0m
[1B][0;32mI (392) cpu_start: ELF file SHA256:  06a34ace76b991bd...[1B][0m
[1B][0;32mI (398) cpu_start: ESP-IDF:          v4.0[1B][0m
[1B][0;32mI (403) cpu_start: Single core mode[1B][0m
[1B][0;32mI (407) heap_init: Initializing. RAM available for dynamic allocation:[1B][0m
[1B][0;32mI (414) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[1B][0m
[1B][0;32mI (420) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[1B][0m
[1B][0;32mI (426) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[1B][0m
[1B][0;32mI (433) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM[1B][0m
[1B][0;32mI (439) heap_init: At 3FFCFE18 len 000101E8 (64 KiB): DRAM[1B][0m
[1B][0;32mI (445) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM[1B][0m
[1B][0;32mI (451) heap_init: At 40078000 len 00008000 (32 KiB): IRAM[1B][0m
[1B][0;32mI (457) heap_init: At 40094950 len 0000B6B0 (45 KiB): IRAM[1B][0m
[1B][0;32mI (464) cpu_start: Pro cpu start user code[1B][0m
[1B][0;32mI (482) spi_flash: detected chip: generic[1B][0m
[1B][0;32mI (483) spi_flash: flash io: dio[1B][0m
[1B][0;32mI (484) cpu_start: Starting scheduler on PRO CPU.[1B][0m
Start Pressing PRG Button if you want to Charge.

[1B][0;32mI (6412) BTDM_INIT: BT controller compile version [79dd6f1][1B][0m
[1B][0;32mI (6413) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[1B][0m
[1B][0;32mI (6531) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0[1B][0m
BT_GAP: Event 10

BT_SPP_Init() => +0

$POGNR,0,0,,+0.0,0,+0,,*6A

TaskGPS:

TaskRF: v12

TaskGPS: 9600bps

TaskSENS: BMP280:  ?!

$POGNR,0,0,,-108.5,4,+33,,*54

$POGNR,0,0,,-108.5,6,+33,,*56

$POGNR,0,0,,-106.0,6,+33,,*5D

$POGNR,0,0,,-104.5,8,+33,,*54

TaskGPS: 19200bps

$POGNR,0,0,,-103.0,10,+33,,*6F

$POGNR,0,0,,-102.5,12,+33,,*69

TaskGPS: 38400bps

$POGNR,0,2,,-102.0,14,+33,,*68

$POGNR,0,2,,-101.5,14,+33,,*6E

TaskGPS: 57600bps

$POGNR,0,2,,-101.0,16,+33,,*69

$POGNR,0,2,,-101.0,18,+33,,*67

TaskGPS: 115200bps

$POGNR,0,2,,-101.0,20,+33,,*6C

$POGNR,0,2,,-101.0,22,+33,,*6E

TaskGPS: 230400bps

$POGNR,0,2,,-101.0,24,+33,,*68

$POGNR,0,5,,-100.5,26,+33,,*69

..........

pjalocha, 2020-10-02:
It is most likely the way the GPS is wired.
How did you assign the GPS signals, did you use the config file which corresponds to this wiring ?

Hidvegi82, 2020-10-02:
Both use the factory NEO 6M (TX-IO34, RX-IO12).
Interestingly, one of the motherboards worked with the same software version before.
[image]
[image]

acasadoalonso, 2020-10-02:
Did you noticed that the boards shown are on different versions ???
When you compile the FW you need to use the proper config file !!!

AC/.

pjalocha, 2020-10-02:
Did you use the T-Beam_v10.cfg thus like this:

cp T-Beam_v10.cfg main/config.h
make
make flash

Hidvegi82, 2020-10-02:
Yes, I noticed. 
The problem is that it previously worked with the same version and the same configuration file.
This OGN (T22_V1.0) worked with the following configuration files for several weeks.
But after picking up the Meshtastic program (which sent the other motherboard) and then reloading the previously used version of the ESP32-OGN tracker, It hasn’t seen the satellite since.
I know this program is not faulty because it used to work with many of these motherboards. 

#define DEFAULT_AcftType        1          // [0..15] default aircraft-type: glider
#define DEFAULT_GeoidSepar     40          // [m]
#define DEFAULT_CONbaud    115200
#define DEFAULT_PPSdelay      100
#define DEFAULT_FreqPlan        0

// #define WITH_HELTEC_V2                        // HELTEC module: PCB LED on GPI025
// #define WITH_HELTEC                        // HELTEC module: PCB LED on GPI025
// #define WITH_TTGO                          // TTGO module: PCB LED on GPIO2, GPIO25 free to use as DAC2 output
// #define WITH_TBEAM                          // T-Beam module
#define WITH_TBEAM_V10                      // T-Beam module
// #define WITH_M5_JACEK                         // JACEK M5 ESP32 OGN-Tracker
// #define WITH_FollowMe                         // by Avionix

// #define WITH_ILI9341                        // 320x240 M5stack
// #define WITH_ST7789                         // IPS 240x240 ST7789
// #define WITH_TFT_LCD                       // TFT LCD
// #define WITH_OLED                          // OLED display on the I2C: some TTGO modules are without OLED display
// #define WITH_OLED2                         // 2nd OLED display, I2C address next higher
#define WITH_U8G2_OLED                     // I2C OLED through the U8g2 library
// #define WITH_U8G2_SH1106
#define WITH_U8G2_LISTS                    // displays various lists, like Relay, Lookout,
// #define U8G2_OLED_PAGES_ENABLED   0b0000001111  // use if WITH_U8G2_OLED is on, 10 pages currently, see end of disp.ccp for the list of pages

#define WITH_RFM95                         // RF chip selection:  both HELTEC and TTGO use sx1276 which is same as RFM95

// #define WITH_SLEEP                         // with software sleep mode controlled by the long-press on the button
// #define WITH_LONGPRESS_SLEEP
#define WITH_CHARGE_MODE  

#define WITH_AXP                           // with AXP192 power controller (T-BEAM V1.0)
// #define WITH_BQ                            // with BQ24295  power controller (FollowMe)

// #define WITH_LED_RX
// #define WITH_LED_TX

// #define WITH_GPS_ENABLE                    // use GPS_ENABLE control line to turn the GPS ON/OFF
#define WITH_GPS_PPS                       // use the PPS signal from GPS for precise time-sync.
#define WITH_GPS_CONFIG                    // attempt to configure higher GPS baud rate and airborne mode

#define WITH_GPS_UBX                       // GPS understands UBX
// #define WITH_GPS_MTK                       // GPS understands MTK
// #define WITH_GPS_SRF
// #define WITH_MAVLINK

// #define WITH_GPS_UBX_PASS                  // to pass directly UBX packets to/from GPS
// #define WITH_GPS_NMEA_PASS                  // to pass directly NMEA to/from GPS

// #define WITH_BMP180                        // BMP180 pressure sensor
#define WITH_BMP280                        // BMP280 pressure sensor
// #define WITH_BME280                        // BMP280 with humidity (but still works with BMP280)
// #define WITH_MS5607                        // MS5607 pressure sensor

#define WITH_PFLAA                         // PFLAU and PFLAA for compatibility with XCsoar and LK8000
// #define WITH_POGNT
#define WITH_LOOKOUT

#define WITH_CONFIG                        // interpret the console input: $POGNS to change parameters

#define WITH_BEEPER                        // with digital buzzer
// #define WITH_SOUND                         // with analog sound produced by DAC on pin 25

// #define WITH_KNOB
// #define WITH_VARIO

// #define WITH_SD                            // use the SD card in SPI mode and FAT file system
// #define WITH_SPIFFS                        // use SPIFFS file system in Flash
// #define WITH_LOG                           // log own positions and other received to SPIFFS and possibly to uSD

#define WITH_BT_SPP                        // Bluetooth serial port for smartphone/tablet link
// #define WITH_WIFI                          // attempt to connect to the wifi router for uploading the log files
// #define WITH_SPIFFS_LOG                    // log transmitted and received packets to SPIFFS

// #define WITH_ENCRYPT                       // Encrypt (optionally) the position

pjalocha, 2020-10-02:
I can only suspect the Meshtastic would set the GPS is some special mode and make it remember this setting ?
Does the GPS starting flashing the PPS after some time when exposed to the sky ?

Hidvegi82, 2020-10-02:
**Does the GPS start flashing the PPS after a while when exposed to the sky?**
Yes, if I don’t use it for a long time, for a few minutes, but if I only restart it right away. So as it should work normally. But only the led ...
The solution will be in the hands of Meshtastic, right?

pjalocha, 2020-10-02:
Maybe it is worth to ask there, open an issue like here, if they are aware of doing some permanent changes to the GPS.
If they can explain this, then we would know how to unlock it.

Hidvegi82, 2020-10-02:
Ok, I'll ask them. Maybe this information may be important later ...
Thanks for the help !!!

Hidvegi82, 2020-10-04:
Hi
I contacted support for the other program.
I quote their answer:
"Meshtastic changes the GPS to run in UBX mode. The other SW needs NMEA mode.
@eriktheV-king has made a small program to reset the mode for the GPS. It is described in this post
[https://meshtastic.discourse.group/t/gps-does-not-work-in-my-application-after-installing-mestastic/1090/7](https://meshtastic.discourse.group/t/gps-does-not-work-in-my-application-after-installing-mestastic/1090/7)"
I uploaded the sketch from the attached link and everything is fine again.
Maybe it helps someone else.
Thank you again for your help!

pjalocha, 2020-10-04:
OK, thank you, I will add a protection against cuch situation in the software.