pjalocha/esp32-ogn-tracker #27: Compling with an RFM69

min0me, 2020-12-14:
Hello,

I’ve seen with the M5 config, that it might be possible to compile the code to build a tracker to use with the RFM69HCW rather then the RFM95, my goal is to straight up use the OGN network rather then via The Things network. I am trying using a generic Weemos Lilon ESP32 as a test bed.

When I type “**make flash**” I get the following error:

main/rf.cpp: In function 'uint8_t StartRFchip()':
/main/rf.cpp:190:7: error: 'class RFM_TRX' has no member named 'OGN_Configure'; did you mean 'FSK_Configure'?
   TRX.OGN_Configure(0, OGN_SYNC);                                // setup RF chip parameters and set to channel #0
       ^~~~~~~~~~~~~
       FSK_Configure
main/rf.cpp: At global scope:
main/rf.cpp:169:13: warning: 'void SetFreqPlanWAN()' defined but not used [-Wunused-function]
 static void SetFreqPlanWAN(void)     

make[1]: *** [/esp-idf/make/component_wrapper.mk:292: rf.o] Error 1
make: *** [/esp-idf/make/project.mk:590: component-main-build] Error 2

I am using version 4.0 of the ESP-IDF and the current branch from the repo which I am building on Debian 10.7 as of writing this.

Thank you in advance

------------------------------
**my config:**

#define DEFAULT_AcftType        1          // [0..15] default aircraft-type: glider
#define DEFAULT_GeoidSepar     40          // [m]
#define DEFAULT_CONbaud    115200
#define DEFAULT_PPSdelay      100
#define DEFAULT_FreqPlan        0

// #define WITH_HELTEC                        // HELTEC module: PCB LED on GPI025
// #define WITH_TTGO                          // TTGO module: PCB LED on GPIO2, GPIO25 free to use as DAC2 output
// #define WITH_TBEAM                          // T-Beam module
// #define WITH_TBEAM_V10                      // T-Beam module
// #define WITH_JACEK                         // JACEK ESP32 OGN-Tracker
#define WITH_M5_JACEK                         // JACEK M5 ESP32 OGN-Tracker
// #define WITH_FollowMe                         // by Avionix

//#define WITH_ILI9341                        // 320x240 M5stack
// #define WITH_ST7789                         // IPS 240x240 ST7789
// #define WITH_TFT_LCD                       // TFT LCD
// #define WITH_OLED                          // OLED display on the I2C: some TTGO modules are without OLED display
// #define WITH_OLED2                         // 2nd OLED display, I2C address next higher
// #define WITH_U8G2_OLED                     // I2C OLED through the U8g2 library (TTGO, HELTEC, T-Beam and FollowMe)

#define WITH_RFM69                         // Jacek design uses RFM69
//#define WITH_RFM95                         // RF chip selection:  both HELTEC and TTGO use sx1276 which is same as RFM95

// #define WITH_SLEEP                         // with software sleep mode controlled by the long-press on the button

// #define WITH_AXP                           // with AXP192 power controller (T-BEAM V1.0)
// #define WITH_BQ                            // with BQ24295  power controller (FollowMe)

// #define WITH_LED_RX
// #define WITH_LED_TX

// #define WITH_GPS_ENABLE                    // use GPS_ENABLE control line to turn the GPS ON/OFF
#define WITH_GPS_PPS                       // use the PPS signal from GPS for precise time-sync.
#define WITH_GPS_CONFIG                    // attempt to configure higher GPS baud rate and airborne mode

 #define WITH_GPS_UBX                       // GPS understands UBX
//#define WITH_GPS_MTK                       // GPS understands MTK
// #define WITH_GPS_SRF
// #define WITH_MAVLINK

// #define WITH_GPS_UBX_PASS                  // to pass directly UBX packets to/from GPS
// #define WITH_GPS_NMEA_PASS                  // to pass directly NMEA to/from GPS

// #define WITH_BMP180                        // BMP180 pressure sensor
// #define WITH_BMP280                        // BMP280 pressure sensor
#define WITH_BME280                        // BMP280 with humidity (but still works with BMP280)
// #define WITH_MS5607                        // MS5607 pressure sensor

// #define WITH_FANET
//#define WITH_PFLAA                         // PFLAU and PFLAA for compatibility with XCsoar and LK8000
// #define WITH_POGNT
#define WITH_LOOKOUT

//#define WITH_CONFIG                        // interpret the console input: $POGNS to change parameters

// #define WITH_BEEPER                        // with digital buzzer
// #define WITH_SOUND                         // with analog sound produced by DAC on pin 25

// #define WITH_KNOB
// #define WITH_VARIO

// #define WITH_SD                            // use the SD card in SPI mode and FAT file system
#define WITH_SPIFFS                        // use SPIFFS file system in Flash
#define WITH_LOG                           // log own positions and other received to SPIFFS and possibly to uSD

//#define WITH_BT_SPP                        // Bluetooth serial port for smartphone/tablet link
// #define WITH_WIFI                          // attempt to connect to the wifi router for uploading the log files
// #define WITH_SPIFFS_LOG                    // log transmitted and received packets to SPIFFS

//#define WITH_ENCRYPT                       // Encrypt (optionally) the position

#define WITH_AP
#define WITH_HTTP

pjalocha, 2020-12-14:
It should be fixed now.
WITH_LORAWAN will not work.
It could in principle, but needs some work.

min0me, 2020-12-22:
Hello,

Thank you for taking a look at that and fixing it.
It compiles OK now

closed 2020-12-22 by min0me

Paulie92, 2021-01-18:
Hello,

I have also find out that I cannot build the code with "WITH_LORAWAN", but in my case I also cannot build the code with "WITH_WIFI" or "WITH_AP".

Should this work or is there also some known bug in the code?

Thanks.

pjalocha, 2021-01-18:
LoRaWAN needs the LoRa chip, this is why. In principle, LoRaWAN has an FSK channel but the code for this is not yet there, it would need some work.
It should not have influence on WIFI or AP but I did not compile it like this for long time, thus I probably forgot something, I will check.

Paulie92, 2021-01-18:
I am compiling the code with "WITH_RFM95 ", the error log is here:

atic library esp-idf\main\libmain.a
d18/20] Generating ld/esp32.project.l
[19/20] Linking CXX executable app-template.elf
FAILED: app-template.elf 
cmd.exe /C "cd . && C:\Users\Paulie\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\app-template.elf.rsp  -o app-template.elf  && cd ."
c:/users/paulie/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(disp_oled.cpp.obj):(.literal._Z18OLED_DrawStatusBarP11u8g2_structP12GPS_Position+0x30): undefined reference to `WIFI_isConnected()'
c:/users/paulie/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(disp_oled.cpp.obj): in function `OLED_DrawStatusBar(u8g2_struct*, GPS_Position*)':
C:\Pavel\Hobby\Arduino\ESP32\OGN_Tracker\esp32-ogn-tracker-master_verze_21_01_18\esp32-ogn-tracker-master\build/../main/disp_oled.cpp:579: undefined reference to `WIFI_isConnected()'
c:/users/paulie/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(hal.cpp.obj):(.literal._Z15CONS_UART_Writec+0x0): undefined reference to `AP_Write(char)'
c:/users/paulie/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(hal.cpp.obj): in function `CONS_UART_Write(char)':
C:\Pavel\Hobby\Arduino\ESP32\OGN_Tracker\esp32-ogn-tracker-master_verze_21_01_18\esp32-ogn-tracker-master\build/../main/hal.cpp:850: undefined reference to `AP_Write(char)'
c:/users/paulie/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.cpp.obj):(.literal.app_main+0xa0): undefined reference to `vTaskAP'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

There is my configuration:
#define DEFAULT_AcftType        1          // [0..15] default aircraft-type: glider
#define DEFAULT_GeoidSepar     40          // [m]
#define DEFAULT_CONbaud    115200
#define DEFAULT_PPSdelay      100
#define DEFAULT_FreqPlan        0
#define WITH_FollowMe                         // by Avionix
#define WITH_U8G2_OLED                     // I2C OLED through the U8g2 library
#define WITH_U8G2_SH1106                   // correct controller for the bigger OLED
#define WITH_U8G2_FLIP                     // flip the OLED screen (rotate by 180deg)
#define WITH_RFM95                         // RF chip selection:  both HELTEC and TTGO use sx1276 which is same as RFM95
#define WITH_GPS_ENABLE                    // use GPS_ENABLE control line to turn the GPS ON/OFF
#define WITH_GPS_PPS                       // use the PPS signal from GPS for precise time-sync.
#define WITH_GPS_CONFIG                    // attempt to configure higher GPS baud rate and airborne mode
#define WITH_GPS_MTK                       // GPS understands MTK
#define WITH_BME280                        // BMP280 with humidity (but still works with BMP280)
#define WITH_FANET                         // FANET transmission and reception
#define WITH_PFLAA                         // PFLAU and PFLAA for compatibility with XCsoar and LK8000
#define WITH_LOOKOUT
#define WITH_CONFIG                        // interpret the console input: $POGNS to change parameters
#define WITH_SPIFFS                        // use SPIFFS file system in Flash
#define WITH_SPIFFS_FAT
#define WITH_LOG                           // log own positions and other received to SPIFFS
#define WITH_BT_SPP                        // Bluetooth serial port for smartphone/tablet link
#define WITH_WIFI                          // attempt to connect to the wifi router for uploading the log files
#define WITH_AP

The compilation is OK only when I comment out the WIFI and AP.

pjalocha, 2021-01-18:
I added:
#define WITH_HTTP
and it compiled.
But it is actually with RFM95 thus you could use WITH_LORAWAN.

Paulie92, 2021-01-18:
I added also #define WITH_HTTP but it didn't helped for me. Still the same problems.

pjalocha, 2021-01-18:
OK, I check it it up to date with the repository, maybe ESP-IDF version, I moved a bit to v4.1
Did you try git pull if there are any differences ?