pjalocha/esp32-ogn-tracker #28: Oled SSH1106 with BMP280 on i2c over T-BEAM v1.1 sx1276-N6M Display don’t start.

Fab501, 2020-12-23:
Dear Pawel,

I have just installed you OGN-tracker to work with my Stratux on a T-BEAM and that works fine from firmware.binary coming from new Stratux eu022. 

I have then compiled the firmware with WITH_U8G2_OLED WITH_U8G2_SH1106 WITH_BMP280 and I have connected both BMP280 and Oled 1.3 SH1106 to i2c pins 3V3,GND,21,22... but most of the time to not say always the display don't bright up. 

But in rare case it start and give information like that Bmp280 is working. 

I have found that if I power the Oled display from 5V and let BMP280 to the 3V3 there no problem anymore.

Could it be a problem with AXP192 ? 

And last thing, using the button I can select the page with all data from BMP280... but data are most of the time void... in fact data are refresh during about 2-3seconds then disappear for near to 7-10sec then reappear and so on ... and on the web page the status page give BMP280 information only if I refresh during the moment where data are on the display...

Any idea ?

Thanks a lot 

Regards

pjalocha, 2020-12-23:
The OLED troubles could be power-up and initialization issues: there is no explicit reset line for the OLED thus depending how it comes up, can change things: possibly putting a longer wait before starting the OLED could help thus look around line 1851 of hal.cpp and maybe add more delay before starting the OLED ?

Just before that is the AXP192 startup thus putting more delay between AX192 turning on the supplies and the OLD starting could help.

For the BMP280 I can explain it: it has to do with the GPS update rate.
The code was written with the 1Hz update rate in mind and Stratux likes higher rates for its AHRS sensors.
The BMP280 is read at fixed rate 2Hz and merging its data with the GPS which runs at a different rate is probably causing this effect. I will work on this.

Nevertheless Stratux should still get pressure updates at 2Hz as sending out the $POGNB does not depend on the GPS.

Fab501, 2020-12-24:
Thanks Pauwel

I have try to add vTaskDelay(1000); with a ifdef WITH_U8G2_SH1106 but that does not work.

I have try with Oled SsD1306 no problem and with SH1106 alone without BMP280 and connected to pin 3V3 it seems that the display start to bright only when powered by battery not when powered by USB... or if I power with 5V in place of 3V3 that should come from AXP192...

Regards

Fab501, 2020-12-24:
For the Display of information from BMP280 you are wright reducing to 1Hz resolve the lack of refresh 👍

pjalocha, 2020-12-25:
Please try the newest code,the pressure data should be fixed there. Please let me know how it works for you.

Fab501, 2020-12-26:
Hi Pauwel,
Thanks for your work,
I have git clone Esp32-OGNTracker and reflash with your changes.

I have soldered Oled SH1106 with BARO BMP280 to avoid defect contacts 

Like before when I power from 3V3 the display don't start but BARO works and NMEA from UART give $GPRMZ with altitude.

Stratux receive well the pressure altitude too.

But the OGNTracker website don't give anymore the values from BARO never...

With power from 5V display start and display  that BARO are At 0x76 but on the BARO page there are no value never or near to never I have one time see some values that have disappear in the next second...

I have test with refresh of 1Hz and refresh of 5Hz. Do not change anything...

Finally I have reflash WiTH_BMP280 in place of BME280 no change neither.

The battery icône flash every second so I suppose that value are refresh also...

Then I put the unit outside to try having a Fix to see if that change something and as soon as the unit receive a time I suppose from one satellite ?!? It immediately display all values 😁 

So it seems that the values can't be displayed without the satellite time. 

My GNSS Mode is set to 0x67 for Stratux if is better to send Galileo and SBAS should I change the GNSS mode to send SBAS and Galileo to Stratux ?

Thanks Regards

Fab501, 2020-12-26:
Is there a way to display Pressure Altitude in Feet and Vario in Feet/min ? 
Thanks

pjalocha, 2020-12-27:
The display of the pressure data should be there even without a GPS lock.
I suspect by catching the lock the GPS update rate went down, thus this is still the problem.
I have made another iteration to approach this, please try it the current code.

If you want to display pressure altitude and vario in feet you need to rewrite the code or maybe create another page with feets instead of meters ? I could do it when I find a bit of time, should not be difficult. What using the speed should be: knots ?

BME280 code should be good for both BMP280 and BME280, it recognizes the sensor type as well recognizes absence of the sensor this WITH_BME280 can be there always, regardless if you have the sensor or not.

Fab501, 2020-12-27:
Dear Pauwel thanks a lot,

In fact the display don't need a gps lock but wait to have the GPS time...

I will check your new code tomorrow and let you know...

I will try to make a new page copy paste of yours so that we can choice between feet/min, feet for pressure altitude and for speed that dépend of the aircraft soms are in kph or knots sometimes mph... For me kph is good because it is just a ground speed so less useful than IAS but we need a second BARO to calculate AirSpeed IAS.

Thanks Regards

Fab501, 2020-12-28:
Dear Pawel,

That seems to works fine now.
Indoor or Outdoor.

After first flash the display don't give time and so no BARO value neither.

I goes outdoor and immediately time and BARO comes on display in seconds...

After some minutes I received the gps fix and goes indoor and remove power and try multiple power on and each time the unit display immediately the time and the BARO values 👍

I don't know if the unit store the time after the first time I goes outdoor but now even if it stay OFF 5 minutes and if I power  ON indoor the time goes immediately 👌

The problem seems corrected great thanks for your job.

I have also test with BME280 I have just received.

If weather are ok I will try this week in plane with Stratux.

The only think that I can not explain is why does the display don't start when connected with BARO to 3V3 power and all goes Wright when powered by 5V but it is not a problem finally 😁

Thanks 

Regards

Fab501, 2020-12-29:
Hi Pawel,

Just to confirm that after a night power off when power on the time is not set and indoor  due to the cellular concrete house there are no satellite signal or very few... after 5 minutes waiting I have seen only a second the BARO values.

I put the device outdoor and in less than 5 seconds the time was set and the BARO values are immediately displayed and refreshed every seconds.

So the GPS fix is not required BUT the time need to be set by satellite signal so that the BARO values are displayed.

In fact it is not a real problem if we consider that the device need to be used with satellite reception. 

For me no need for futur investigation but I stay available if you want to make more tests.

Thanks for your efforts and great software 👍

Regards

Fab501, 2021-01-01:
Hi Pawel 

Happy new Year 

Just to give last information, I have receive last version of T-BEAM v1.1 from AliExpress SX1262 and M8N and pigtail Antenna AND I have no more problem with Oled that don't start when connected to 3V3.

So problem was hardware...

Thanks Regards

pjalocha, 2021-01-01:
Thank you for letting me and others know.
It is worth noticing there are those different T-Beam versions and the GPS can as well be different even on same version of the T-Beam, thus like v1.0 or v1.1
The T-Beam with M8N is clearly an interesting option, I hear the GPS there is fairly good and can receive several GNSS systems, catching like 24 satellites is possible. The code here does not support yet enabling those systems but it soon should.

Fab501, 2021-01-01:
Yes both versions I have ordered was v1.1 T-BEAM Lilygo but the first use a fix SMA connector for antenna a GPS M6N and a Lora32 with SX1276 and the new one use pigtail antenna SX1262 and M8N but there are no differences printed on the board to distinct both set of parts all are V1.1 and finally there are some differences since same software start the Oled 1.3 SH1106 and the second do not on 3V3. 
Regards

bortek, 2021-01-09:
Thanks for the update. I am to order my the latest board that you mention with M8N and SX1262.

Fab501, 2021-01-10:
Hi Bortek,

Here the Config I used with T-BEAM SX1262-M8N and OLED 1.3" SH1106 and BME280 and Bluetooth and STRATUX you can adapt for your need:

#define DEFAULT_AcftType        8          // [0..15] default aircraft-type: Powered Aircraft
#define DEFAULT_GeoidSepar     40          // [m]
#define DEFAULT_CONbaud    115200
#define DEFAULT_PPSdelay      100
#define DEFAULT_FreqPlan        0

// #define WITH_HELTEC                        // HELTEC module: PCB LED on GPI025
// #define WITH_HELTEC_V2                     // HELTEC module v2
// #define WITH_TTGO                          // TTGO module: PCB LED on GPIO2, GPIO25 free to use as DAC2 output
// #define WITH_TBEAM                          // T-Beam module
#define WITH_TBEAM_V10                      // T-Beam module
// #define WITH_M5_JACEK                         // JACEK M5 ESP32 OGN-Tracker
// #define WITH_FollowMe                         // by Avionix

// #define WITH_ILI9341                        // 320x240 M5stack
// #define WITH_ST7789                         // IPS 240x240 ST7789
// #define WITH_TFT_LCD                       // TFT LCD
// #define WITH_OLED                          // OLED display on the I2C: some TTGO modules are without OLED display
// #define WITH_OLED2                         // 2nd OLED display, I2C address next higher
#define WITH_U8G2_OLED                     // I2C OLED through the U8g2 library
#define WITH_U8G2_SH1106                   // correct controller for the bigger OLED
// #define WITH_U8G2_FLIP                     // flip the OLED screen (rotate by 180deg)

#define WITH_RFM95                         // RF chip selection:  both HELTEC and TTGO use sx1276 which is same as RFM95

// #define WITH_SLEEP                         // with software sleep mode controlled by the long-press on the button

#define WITH_AXP                           // with AXP192 power controller (T-BEAM V1.0)
// #define WITH_BQ                            // with BQ24295  power controller (FollowMe)

// #define WITH_LED_RX
// #define WITH_LED_TX

// #define WITH_GPS_ENABLE                    // use GPS_ENABLE control line to turn the GPS ON/OFF
#define WITH_GPS_PPS                       // use the PPS signal from GPS for precise time-sync.
#define WITH_GPS_CONFIG                    // attempt to configure higher GPS baud rate and airborne mode

#define WITH_GPS_UBX                       // GPS understands UBX
// #define WITH_GPS_MTK                       // GPS understands MTK
// #define WITH_GPS_SRF
// #define WITH_MAVLINK

#define WITH_GPS_UBX_PASS                  // to pass directly UBX packets to/from GPS
#define WITH_GPS_NMEA_PASS                  // to pass directly NMEA to/from GPS

// #define WITH_BMP180                        // BMP180 pressure sensor
// #define WITH_BMP280                        // BMP280 pressure sensor
#define WITH_BME280                        // BMP280 with humidity (but still works with BMP280)
// #define WITH_MS5607                        // MS5607 pressure sensor
// #define WITH_MS5611                        // MS5611 pressure sensor

// #define WITH_BMX055                        // BMX055 magnetic and IMU sensor

// #define WITH_LORAWAN                       // LoRaWAN connectivity
// #define WITH_FANET                         // FANET transmission and reception
#define WITH_PAW			   // Add PAW transmission

#define WITH_PFLAA                         // PFLAU and PFLAA for compatibility with XCsoar and LK8000
// #define WITH_POGNT
// #define WITH_GDL90
// #define WITH_PGAV5
#define WITH_LOOKOUT

#define WITH_CONFIG                        // interpret the console input: $POGNS to change parameters

#define WITH_BEEPER                        // with digital buzzer
// #define WITH_SOUND                         // with analog sound produced by DAC on pin 25

// #define WITH_KNOB
// #define WITH_VARIO

// #define WITH_SD                            // use the SD card in SPI mode and FAT file system
#define WITH_SPIFFS                        // use SPIFFS file system in Flash
// #define WITH_SPIFFS_FAT
#define WITH_LOG                           // log own positions and other received to SPIFFS
// #define WITH_SDLOG                         // log own position and other data to uSD card

#define WITH_STRATUX
#define WITH_BT_SPP                        // Bluetooth serial port for smartphone/tablet link
#define WITH_WIFI                          // attempt to connect to the wifi router for uploading the log files
// #define WITH_AP                            // Open Access Point MOde
#define WITH_HTTP                           // Open Web Interface

// #define WITH_ENCRYPT                       // Encrypt (optionally) the position

Hope that help, regards.

Fab501, 2021-01-10:
Just in case, my second board have a little difference in behavior exact same T-BEAM-SX1262-M8N exact same BME280 and OLED SH1106, but when I start the board from microUSB the display don't start like before, but with lithium Battery 3,7V the display start well.

I have found that when I plug usb without battery, if I power off the T-BEAM via the PWR Button next to microUSB holding pressed some seconds, the board shutoff and when I restart via the PWR Button then the display start well, so their are probably something in the init that don't work when plugging USB and that are reset when powering via the PWR Button...Not a big problem :)