pjalocha/esp32-ogn-tracker #31: OGN tracker with Stratux Europe

kaelsaes, 2021-01-17:
Hi,

thanks for your great project.

I'm using your OGN tracker together with my Stratux Europe traffic awareness device.

I love the additional option to send P3I (PilotAware) signals.

My PilotAware Rosetta receives the P3I signals from Stratux Europe / OGN tracker.
But the signals are sent to rare. Sometimes Rosetta has to wait 60 seconds or more to receive a P3I signal even though both devices are placed next to each other.

Do I have an issue with my OGN tracker or is this normal operation?

Thanks in advance.

Best regards
Kai

pjalocha, 2021-01-17:
It is most likely that the device is not moving.
When moving more than 1m/s the OGNTP packet should be sent at 1Hz rate and the PilotAware packet at about 1/4Hz.
When not moving the rates are about 4 times lower.
This is all still subject to optimization, basically the rules of 1% duty must be kept and the OGNTP packets being 5ms, P3I being 8ms long, then there is LoRaWAN which is transmitted at about 1/60Hz and packets are about 60ms long, and yet FANET packets, at about 1/20Hz rate 30ms long.
If you can, please check what happens when the OGN-Tracker is moving more than 1m/s.

kaelsaes, 2021-01-17:
I have tested last week in my car. The problem still appeared.
I will test in flight when the weather becomes better.
Is FANET already available?

pjalocha, 2021-01-17:
Please send me the Stratux .ogn log files from your car test via the private channel.
I can then see how often the P3I position was transmitted.

kaelsaes, 2021-01-18:
I have sent the log files.

Thanks