pjalocha/esp32-ogn-tracker #32: Can't get BMP280 to work on T-Beam V0.7

Shredder, 2021-01-28:
I am unable to get my BMP280 working with my T-Beam V0.7. I have compiled the latest revision with `WITH_BMP280` and separately with `WITH_BME280`, but neither worked. Otherwise the T-Beam works fine.

I have tried the following PIN connections:

VCC -> 3V3
GND -> GND
SCL -> 22
SDA -> 21

and also:

VCC -> 5V
GND -> GND
SCL -> 2
SDA -> 13

On Minicom, I see a message similar `TaskSENS: BMP280:  ?!`, which I assume means that no BMP280 could be found. I have also tried without connecting the OLED display in both positions, but no change.

I have tried In both cases, I also connected my small OLED display to PINs 3V3 / GND / 22 / 21. The OLED display works, but in the BMP280 screen, all the fields are blank.

Here is what my BMP280 looks like. Do you see any obvious issues (is it a supported BMP280? is the PIN order as expected?)?

Any suggestion is appreciated.

[image]
[image]

Here is my `config.h`:

```
#define DEFAULT_AcftType        8          // [0..15] default aircraft-type: glider
#define DEFAULT_GeoidSepar     40          // [m]
#define DEFAULT_CONbaud    115200
#define DEFAULT_PPSdelay      100
#define DEFAULT_FreqPlan        0

// #define WITH_HELTEC                        // HELTEC module: PCB LED on GPI025
// #define WITH_TTGO                          // TTGO module: PCB LED on GPIO2, GPIO25 free to use as DAC2 output
#define WITH_TBEAM                          // T-Beam module
// #define WITH_TBEAM_V10                      // T-Beam module
// #define WITH_JACEK                         // JACEK ESP32 OGN-Tracker
// #define WITH_M5_JACEK                         // JACEK M5 ESP32 OGN-Tracker
// #define WITH_FollowMe                         // by Avionix

// #define WITH_ILI9341                        // 320x240 M5stack
// #define WITH_ST7789                         // IPS 240x240 ST7789
// #define WITH_TFT_LCD                       // TFT LCD
// #define WITH_OLED                          // OLED display on the I2C: some TTGO modules are without OLED display
// #define WITH_OLED2                         // 2nd OLED display, I2C address next higher
#define WITH_U8G2_OLED                     // I2C OLED through the U8g2 library

#ifdef WITH_JACEK
#define WITH_RFM69                         // Jacek design uses RFM69
#else
#define WITH_RFM95                         // RF chip selection:  both HELTEC and TTGO use sx1276 which is same as RFM95
#endif

// #define WITH_SLEEP                         // with software sleep mode controlled by the long-press on the button

// #define WITH_AXP                           // with AXP192 power controller (T-BEAM V1.0)
// #define WITH_BQ                            // with BQ24295  power controller (FollowMe)

// #define WITH_LED_RX
// #define WITH_LED_TX

// #define WITH_GPS_ENABLE                    // use GPS_ENABLE control line to turn the GPS ON/OFF
// #define WITH_GPS_PPS                       // use the PPS signal from GPS for precise time-sync.
#define WITH_GPS_CONFIG                    // attempt to configure higher GPS baud rate and airborne mode

#define WITH_GPS_UBX                       // GPS understands UBX
// #define WITH_GPS_MTK                       // GPS understands MTK
// #define WITH_GPS_SRF
// #define WITH_MAVLINK

#define WITH_GPS_UBX_PASS                  // to pass directly UBX packets to/from GPS
#define WITH_GPS_NMEA_PASS                  // to pass directly NMEA to/from GPS

// #define WITH_BMP180                        // BMP180 pressure sensor
#define WITH_BMP280                        // BMP280 pressure sensor
// #define WITH_BME280                        // BMP280 with humidity (but still works with BMP280)
// #define WITH_MS5607                        // MS5607 pressure sensor

#define WITH_PFLAA                         // PFLAU and PFLAA for compatibility with XCsoar and LK8000
// #define WITH_POGNT
#define WITH_LOOKOUT

#define WITH_CONFIG                        // interpret the console input: $POGNS to change parameters

#define WITH_BEEPER                        // with digital buzzer
// #define WITH_SOUND                         // with analog sound produced by DAC on pin 25

// #define WITH_KNOB
// #define WITH_VARIO

// #define WITH_SD                            // use the SD card in SPI mode and FAT file system
#define WITH_SPIFFS                        // use SPIFFS file system in Flash
#define WITH_LOG                           // log own positions and other received to SPIFFS and possibly to uSD

#define WITH_STRATUX
#define WITH_BT_SPP                        // Bluetooth serial port for smartphone/tablet link
#define WITH_WIFI                          // attempt to connect to the wifi router for uploading the log files
// #define WITH_SPIFFS_LOG                    // log transmitted and received packets to SPIFFS
#define WITH_HTTP

//#define WITH_ENCRYPT                       // Encrypt (optionally) the position

#define WITH_PAW
#define WITH_LORAWAN

// #define WITH_SDLOG
// #define WITH_AP
```

Fab501, 2021-01-28:
Dear Shredder,

I don't have a TBEAM v7 but some v11 and I have also have some problems at first try...

If you have a chip named AXP192 you probably should add WITH_AXP.

Then from my test the Oled display 1.3 with chipset SH1106 and a BARO BMP280 was working  when I connect 5V,GND,21,22 with some TBEAM I need to plug a battery so that the display start.

I am pretty sure that the pins 2,13 are not used with OGNTRACKER firmware to be tested.

You can see remotely via Wifi that BMP280 is working using WITH_HTTP WITH_WIFI WITH_AP if I remember correct in the bottom of the main/config.h.

Hope that help

pjalocha, 2021-01-28:
It does work for me for 0.7, the pins are defined such that you solder the BMP280 on four consecutive pins, one-to-one, thus just on four header pins. thus SCL=22 and SDA=21
Just in case define WITH_BME280 and this will handle both BME280 and BMP280

Shredder, 2021-01-30:
Thanks for the responses, I've tried the suggestions and still no luck. I am beginning to think that my BMP280 is faulty. I will order another one and report back.

Shredder, 2021-02-03:
While I am waiting for a replacement BMP280, I am wondering whether this could be an issue of a non-standard I2C address of my sensor. I have seen reports that some cheap sensors apparently have a fixed non-standard I2C address, and the software needs to be configured accordingly.

Could this be the case here?

Shredder, 2021-06-07:
The replacement BMP280 is recognized by the firmware. So the issue was probably just a faulty sensor.

closed 2021-06-07 by Shredder

DoudouManyl, 2023-05-20:
Did you use pull-up resistance?
I have tried with wemos r1d2 , didn't work until I used pull-up resistance.
Hope it will help

pjalocha, 2023-05-20:
For BMP280 I did not need any pullups - did you activate the internal pull-ups by software or you had to use external ones ?

DoudouManyl, 2023-06-03:
> For BMP280 I did not need any pullups - did you activate the internal pull-ups by software or you had to use external ones ?

yes i tried with internal pullup resistors , did not work until i used external pull up resistors.
there was a little confusion with my sensor , it was actually a bmp280 5V, and it looks like bme280 , although it worked without humidity values . when i swapped to my second sensor wich was the real bme280 , the humidity worked fine .
now i am having problem with the altitude , when i try different exemples from different libraries, it gives me different altitude.
if someone can help ill be glad 
i hope that helped

pjalocha, 2023-06-03:
What i the problem with altitude: the conversion from pressure to pressure altitude ?

DoudouManyl, 2023-06-03:
It gives different reading for each library

pjalocha, 2023-06-03:
Can you give some examples with numbers ?