pjalocha/esp32-ogn-tracker #33: Baro and GPS altitude.

Hidvegi82, 2021-02-25:
Hello
How can I set the baro altitude or reset it before takeoff? (Heltec v2)
Now:
GPS: + 87.3 m
Bmp280: -78.5m (1022.75hPa)

pjalocha, 2021-02-25:
You cannot.
The altitude is Standard Pressure Altitude thus is referred strictly to the static pressure and so depends a lot on the weather.
The difference can be quite a lot, especcially during low pressure fronts.

Hidvegi82, 2021-02-25:
Thanks for the reply.
I understand and I'm sorry if I asked wrong.
Wouldn’t it be possible to have a height value that I reset on the ground (or start from zero when ogn is turned on) and the baro sensor or gps would count the height gained or lost?

pjalocha, 2021-02-25:
There is actually an algorithm running there which attempts to do it automatically, it is just not displayed anywhere.
I will think of making it visible.

Hidvegi82, 2021-02-25:
Thank you very much!!!