pjalocha/esp32-ogn-tracker #34: FANET

lupus78, 2021-03-05:
I am trying to complie for TBEAM v10 with FANET option, but it won't compile. First issue is at rf.ccp:465, there is a typo brek; instead of break; That's an easy fix, but after still this complilation error appears:
`/home/lupus/esp/esp32-ogn-tracker-tbeam/main/hal.cpp: In function 'void CONS_UART_Init()':`
`/home/lupus/esp/esp32-ogn-tracker-tbeam/main/hal.cpp:840:3: sorry, unimplemented: non-trivial designated initializers not supported`

pjalocha, 2021-03-05:
Please use the v4.1 of ESP-IDF
I tried to make the code compile for both but somehow it does not work for 4.0
thus please reinstall the ESP-IDF starting with:
git clone -b v4.1 --recursive https://github.com/espressif/esp-idf.git

lupus78, 2021-03-05:
That Worked. Can I find some explanation on how the FANET implementation works?

And please correct the typo in the source code.

pjalocha, 2021-03-05:
Typo corrected.
What do you mean: how FANET works ?
It sends FANET position message at some interval, fairly low, as those messages are relatively long, and we need to keep the duty cycle.
Or do you ask for FANET protocol description ?
Then see here: https://github.com/3s1d/fanet-stm32/blob/master/Src/fanet/radio/protocol.txt

lupus78, 2021-03-05:
Not the protocol, but more like what your software does. It's a big problem with your code is that it lacks documentation, there are tons of options in the config.h but very minimal comments to be able to understand what does what. For example, if I understand correctly there is a web interface via Wifi to see configuration and download logs.... but I was not able to make it work.

I am happy to help you to make a proper documentation, maybe we could do it under the "Wiki" section here on GitHub