pjalocha/esp32-ogn-tracker #35: Wifi

lupus78, 2021-04-02:
I was able to compile for TBEAM v10 with WITH_AP, WITH_WIFI and WITH_HTTP options. Ctrl+C on monitor shows me the APname and empty APpass but I can't see the wifi network, and also I can not connect to it as a "hidden" wifi network. Am I doing something wrong?

lupus78, 2021-04-03:
I think I found a solution. I have switch up the esp_wifi_set_mode() and esp_wifi_set_config() commans in wifi.cpp. set_mode should come first, but in the code set_config comes first. When I put set_mode before set_config, the wifi started to work.

pjalocha, 2021-04-04:
OK, I believe you, I swapped the two calls, although I did not notice a problem, I could still connect to WiFi.
BTW: I added WITH_AP_BUTTON which makes the AP start only when the user button is pressed on startup,
This saves about 80mA when WiFi is not required.
I am not sure how frequent restarts affect the internal flash file system, this should be watched.

lupus78, 2021-04-04:
That's a nice idea. I was thinking about a "Parameter" to turn the Wifi on or off, and the user could control it with Bluetooth.

However, I think a nice Bluetooth app for smartphone would be the best way to configure the device.

lupus78, 2021-04-04:
Ah, and some visual indication on the screen which shows if the Wifi is ON!

pjalocha, 2021-04-04:
You can control it already: if you set APname to empty, then WiFi AP would not start.

I actually am not sure now, because any unexpected restart would turn the WIFi AP off thus not sure if this is good is some circumstances.

pjalocha, 2021-04-04:
For the indication, I will make one, as there is already a WiFi icon there, just not properly controlled yet.

lupus78, 2021-04-06:
Setting the APname empty seams to be an odd solution to me. A dedicated on/off parameter is better.