pjalocha/esp32-ogn-tracker #43: can't make it work with GN-801

qubolino, 2022-02-13:
i have a ttgo lora board i'm trying to enhance with the GN-801 GPS. i can't seem to get it work although i do get data with the following snippet in arduino
```
) {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial2.begin(9600, SERIAL_8N1, 12, 35);

}

void loop() {
  while (Serial2.available()) {
    Serial.print(char(Serial2.read()));
  }
}
```

my wiring is as such
```
#define PIN_GPS_TXD  GPIO_NUM_12
#define PIN_GPS_RXD  GPIO_NUM_35 
#define PIN_GPS_PPS  GPIO_NUM_34
```
and i use the `WITH_TTGO` flag

pjalocha, 2022-02-14:
Which board it is ?
So you don't see any GPS data on the console and only get messages about switching the GPS serial port baudrate ?
I can think of swapped TX/RX or a conflict with another peripherial using the same pin, thus please tell me when board you have, maybe there is a conflict.

qubolino, 2022-02-14:
it's a TTGO Lora with no version number written. I bought it in spring 2020.

indeed i don't see any GPS data on the console and only get messages about switching the GPS serial port baudrate 

i don't think it's swapped TX RX since i do see data with my arduino snippet

wiring is as such:
GN-801 | TTGO
GND | GND | black
VCC | 5V | red
TXD | 12 | green
RXD | 35 | white
PPS | 34 | blue
EN | NC | yellow

qubolino, 2022-02-14:
also tried with pin 14

GN-801 | TTGO | color
-- | -- | --
TXD | 14 | green

`#define PIN_GPS_TXD  GPIO_NUM_14`

works with simple snippet but not ogn-tracker

pjalocha, 2022-02-14:
If you compile WITH_TTGO

#if defined(WITH_HELTEC) || defined(WITH_HELTEC_V2) || defined(WITH_TTGO)
#define PIN_RFM_RST  GPIO_NUM_14  // Reset
#define PIN_RFM_IRQ  GPIO_NUM_26  // packet done on receive or transmit
#define PIN_RFM_SS   GPIO_NUM_18  // SPI chip-select
#define PIN_RFM_SCK  GPIO_NUM_5   // SPI clock
#define PIN_RFM_MISO GPIO_NUM_19  // SPI MISO
#define PIN_RFM_MOSI GPIO_NUM_27  // SPI MOSI
#endif // HELTEC TTGO

so pin 14 is used for RF chip reset

and for the GPS pins:

#if defined(WITH_HELTEC) || defined(WITH_TTGO)
                                  // VK2828U   GN-801   MAVlink
#define PIN_GPS_TXD  GPIO_NUM_13  // green     green    green
#define PIN_GPS_RXD  GPIO_NUM_35 // blue      yellow   yellow
#define PIN_GPS_PPS  GPIO_NUM_34 // white     blue

Could you try this pinout ?

The defines are in main/hal.cpp

qubolino, 2022-02-14:
these are used also 

#define PIN_I2C_SCL GPIO_NUM_15   // SCL pin
#define PIN_I2C_SDA GPIO_NUM_4    // SDA pin
#define OLED_I2C_ADDR 0x3C        // I2C address of the OLED display
#define PIN_OLED_RST GPIO_NUM_16  // OLED RESET: low-active

qubolino, 2022-02-14:
tried pin 23 (12 and 13 may have been slightly overheated), still no success. where can i print in ogn-tracker whatever is read from `PIN_GPS_TXD`
happy to insert snippet recompile and flash

qubolino, 2022-02-14:
no. like i said i think the pin is faulty

pjalocha, 2022-02-14:
Pin 13 is to transmit data to the GPS, it is not essenital.
Pin 35 is the important one, because it receives data from the GPS.
Try to use pin 37 - it is free on TTGO module

Change it in in line 358 in hal.cpp and recompile

#define PIN_GPS_RXD  GPIO_NUM_37 // blue      yellow   yellow

qubolino, 2022-02-14:
> Pin 35 is the important one, because it receives data from the GPS.
i've got my wiring wrong then (gps data comes from gps tx to ttgo 23) and i defined 
```
#define PIN_GPS_TXD  GPIO_NUM_23 // green     green    green
#define PIN_GPS_RXD  GPIO_NUM_35 // blue      yellow   yellow
```

should be the other way around?

qubolino, 2022-02-14:
here's my new definition
```
#define PIN_GPS_TXD  GPIO_NUM_35 // green     green    green
#define PIN_GPS_RXD  GPIO_NUM_37 // blue      yellow   yellow
#define PIN_GPS_PPS  GPIO_NUM_34 // white     blue
```
and wiring
|gps|ttgo|
|-|-|
|TXD|35|
|RXD|37|
|PPS|34|

pjalocha, 2022-02-14:
PIN_GPS_TXD is where the CPU sends data to the GPS
PIN_GPS_RXD is where the GPS sends data to the CPU

BTW: pin 36 is used to measure the battery voltage (althoughnot always conencted).

qubolino, 2022-02-14:
i ordered a new board, but to make things clear
i) my GN-801 has a `TX` pin and NMEA comes out of it.
currently:
`GN-801 TX --NMEA--> TTGO GPIO_NUM_35` and this works (in the sense that with arduino   `Serial2.begin(9600, SERIAL_8N1, 35, 37);` i see GPS data even with pin 37 disconnected
takeaway: CPU gets incoming data on pin 35.

according to this
> PIN_GPS_RXD is where the GPS sends data to the CPU
i should define  `#define PIN_GPS_RXD  GPIO_NUM_35`
do you confirm?

ii) can you advise a line in the code where i can output all traffic incoming from `GPIO_NUM_35` to give another attempt at debugging pls?

qubolino, 2022-02-17:
was in fact a board issue.