pjalocha/esp32-ogn-tracker #44: using as basestation

qubolino, 2022-03-04:
is there a way to compile the code to use this as an ogn basestation?  if not i’m happy to follow guidelines and implement it

pjalocha, 2022-03-04:
You ask to provide a compiled code ?
If so, what is your hardware ?

qubolino, 2022-03-04:
i have both t-beam and heltec at hand
my point is: if i want to use this codebase as ground station, is it possible? is it only a matter of using `WITH_APRS` and `WITH_WIFI`? or is there some development needed?

qubolino, 2022-03-04:
btw, which protocols will it decode simultaneously?

pjalocha, 2022-03-04:
At this time it only receives the OGN-Tracker protocol. I keep thinking of doingother protocols on the time-switching basis.

Yes, you need to compile it WITH_APRS and WITH_WIFI and then you need to configure the WiFi networks by sending command to the console like:

$POGNS,WIFIname=<network-name>,WIFIpass=<network-password>

it should connect by itself to open networks.
You canspecify up to 10 networks using WIFIname0 to WIFIname9 and WIFIpass0 to WIFIpass9

The feature is new, please report how does it work for you.

VirusPilot, 2022-03-04:
just fyi, here is a project that does exactly that: [https://github.com/roema/Open-Glider-Network-Groundstation](url). I have such a basestation running, works quite nicely ...

pjalocha, 2022-03-04:
OK, What kind of range do you get there ?

My code came from a particular setup when they fly the drones and some reception coverage is needed in the area around the launch area, so I realized the tracker of a drone on the ground could serve as a reciever for drones in the air.

VirusPilot, 2022-03-05:
From a few days uptime, it seems that range is less than a standard Pi/SDR setup. Will provide more shortly ...

closed 2022-03-05 by qubolino