pjalocha/esp32-ogn-tracker #55: error at compile (hal.cpp)

manuelmunozjgit, 2023-08-06:
Hi Pavel,
I'm newbie with ogn tracker.

I'm using latest espidf framework and i'm getting some warnings and the following error at build step:

/root/esp32-ogn-tracker/main/hal.cpp: In function 'void IO_Configuration()':
/root/esp32-ogn-tracker/main/hal.cpp:1610:3: error: 'uart_set_rx_full_threshold' was not declared in this scope
   uart_set_rx_full_threshold(GPS_UART, 24);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~
/root/esp32-ogn-tracker/main/hal.cpp:1610:3: note: suggested alternative: 'uart_set_wakeup_threshold'
   uart_set_rx_full_threshold(GPS_UART, 24);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~
   uart_set_wakeup_threshold
/root/esp32-ogn-tracker/main/hal.cpp:1646:3: warning: missing initializer for member 'i2c_config_t::<anonymous>' [-Wmissing-field-initializers]

Regards
Manuel

animevietsub, 2023-08-06:
Please send all of your error logs, and try to change the `CMakeLists.txt` in the `main` folder to:
```
idf_component_register(SRCS "aero.cpp"
							"aes.c"
							"ap.cpp"
							"aprs.cpp"
							"atmosphere.cpp"
							"bitcount.cpp"
							"bt.cpp"
							"cmac.c"
							"cmacutil.c"
							"ctrl.cpp"
							"disp.cpp"
							"disp_lcd.cpp"
							"disp_oled.cpp"
							"format.cpp"
							"gdl90.cpp"
							"gps.cpp"
							"hal.cpp"
							"http.cpp"
							"imu.cpp"
							"intmath.cpp"
							"knob.cpp"
							"ldpc.cpp"
							"log.cpp"
							"LoRaMacCrypto.c"
							"main.cpp"
							"nmea.cpp"
							"ognconv.cpp"
							"proc.cpp"
							"rf.cpp"
							"sdlog.cpp"
							"sens.cpp"
							"st7789.cpp"
							"stratux.cpp"
							"timesync.cpp"
							"wifi.cpp"
						EMBED_FILES "OGN_logo_240x240.jpg"
                       INCLUDE_DIRS "."
					   INCLUDE_DIRS "../components")

```

acasadoalonso, 2023-08-06:
Manuel,
The main issue is with the ESPIDF framework/toolbox, keeping up with the
latest version is a pain on the a**
I am using the toolchain version esp-2020r2
with no problems.
Cheers.
AC/.

manuelmunozjgit, 2023-08-07:
Gracias Alonso !
regards
Manuel

manuelmunozjgit, 2023-08-07:
> Manuel, The main issue is with the ESPIDF framework/toolbox, keeping up with the latest version is a pain on the a** I am using the toolchain version esp-2020r2 with no problems. Cheers. AC/.
> […](#)
> -- AC/. ***@***.***

gracias Alonso.