pjalocha/esp32-ogn-tracker #56: Support for APX2102

naquaii, 2023-08-12:
Seems newer T-BEAM boards (sometimes calls V1.2 but mine's marked V1.1) have a new PMU chip, APX2102 instead of APX192.

The symptoms are that transmission never starts, it's as if the transmitter isn't "turned on".
When flashing from a firmware of another software that works it works until power cycle after which everything seems fine except no transmission.
I've tried coming from both SoftRF 1.2+ and meshtastic and in both cases transmission works until power cycle after flash.

SoftRF 1.1 behaves the same way but 1.2 and 1.3 works. As far as I can see there's a couple of commits relating to this in last summer.
https://github.com/lyusupov/SoftRF/commits/master?before=96cee5cbd953340611c1e6440f4e3a1f0bf3bb37+385&branch=master&qualified_name=refs%2Fheads%2Fmaster

My guess would be that there's some sort of initialization that needs to happen that's new for the APX2102.

wadoadi, 2023-08-12:
I have the exact same problem, with the same chips, and also a new board...

tfede37, 2024-01-08:
Any news about this issue ?

pjalocha, 2024-01-08:
Did you see my project ogn-tracker ?
It is with platformio and it works on T-Beam-S3.
Not all features yet, but should t least transmit an OGN sgnal.

tfede37, 2024-01-09:
I apologize but i do not understand what you are referring to, i am quite new to all this (github etc.)
You are mentioning the T-Beam-S3 while i was refererring to T-Beam LilyGo with the AXP2102 chip.
I don't know if the S3 is same or close to the Lilygo, i just was wondering if you where still working on it or not because i built my own Stratux Receiver and  unfortunatelly it seems that the LilyGo with the old chip is no longer available.

pjalocha, 2024-01-09:
See this prroject: https://github.com/pjalocha/ogn-tracker
It should work on the T-Beam (not-S3) with the AXP2102
Try the profiles: ttgo-sx1276-tbeam or ttgo-sx1262-tbeam
and if this does not work, please come back.

tfede37, 2024-01-09:
Thank you Pawel for your feedback, forgive me but i have just two more questions:
How do i compile the source code in the SRC folder and how do i upload it to the TTGO ?
Thanks a lot for yoy help !

pjalocha, 2024-01-09:
You need to install PlatformIO - you can install command-line or IDE, I only know about command line.

Check this website how to install it: https://platformio.org/install

Once you install it, you type like:

pio run -e ttgo-tbeam-sx1276 -t upload --upload-port=/dev/ttyACM0

tfede37, 2024-01-10:
Hi, finally manged to get use to platformio, tried both ttgo-sx1276-tbeam and  ttgo-sx1262-tbeam config but none of the 2 is transmittin in any protocol (tried all settings possible through stratux web interface).
To be able to compile and uplaod it i had to change XPowersLib from 0.1.8 to 0.2.1 because of an "integrity error", maybe this is the issue but i don't know ha to handle it.
Just let me know yout thoughts.

pjalocha, 2024-01-10:
OK, so it compiles and loads, can you see any output on the serial port ?
Did you choose the corect RF chip: sx1276 or sx1262 ?

wadoadi, 2024-01-10:
I get an error Error: Package version 0.2.1+sha.db9bd81 doesn't satisfy requirements ^0.1.8 based on PackageMetaData <type=library name=XPowersLib version=0.2.1+sha.db9bd81 spec={'owner': None, 'id': None, 'name': 'XPowersLib', 'requirements': '^0.1.8', 'uri': 'git+https://github.com/lewisxhe/XPowersLib'}

wadoadi, 2024-01-10:
I get the following errors, it's my first time trying to use PIO, this when compiling:
pio run -e ttgo-tbeam-sx1276 -t upload --upload-port=/dev/ttyACM9
Error: Unknown environment names 'ttgo-tbeam-sx1276'. Valid names are 'ttgo-sx1276-tbeam-07, ttgo-sx1262-tbeam-s3, ttgo-sx1276-tbeam-07-stratux, ttgo-sx1276-tbeam, ttgo-sx1262-tbeam'

Error: Package version 0.2.1+sha.db9bd81 doesn't satisfy requirements ^0.1.8 based on PackageMetaData <type=library name=XPowersLib version=0.2.1+sha.db9bd81 spec={'owner': None, 'id': None, 'name': 'XPowersLib', 'requirements': '^0.1.8', 'uri': 'git+https://github.com/lewisxhe/XPowersLib'}

wadoadi, 2024-01-10:
I get the following errors, it's my first time trying to use PIO and is via VS Code, this is when compiling with:
pio run -e ttgo-tbeam-sx1276 -t upload --upload-port=/dev/ttyACM9
Error: Unknown environment names 'ttgo-tbeam-sx1276'. Valid names are 'ttgo-sx1276-tbeam-07, ttgo-sx1262-tbeam-s3, ttgo-sx1276-tbeam-07-stratux, ttgo-sx1276-tbeam, ttgo-sx1262-tbeam'

so then I tried:

pio run -e ttgo-sx1276-tbeam-07-stratux -t upload --upload-port=/dev/ttyACM9

but I now get the following error:

Error: Package version 0.2.1+sha.db9bd81 doesn't satisfy requirements ^0.1.8 based on PackageMetaData <type=library name=XPowersLib version=0.2.1+sha.db9bd81 spec={'owner': None, 'id': None, 'name': 'XPowersLib', 'requirements': '^0.1.8', 'uri': 'git+[https://github.com/lewisxhe/XPowersLib'}](https://github.com/lewisxhe/XPowersLib'%7D)

pjalocha, 2024-01-11:
For me, this command went through, I use Lubuntu on x86_64 laptop.
But yo changed the XPowerLib version and it works now ?

tfede37, 2024-01-11:
wadoladi got the same error i got, it seems we are not able to get the Xpowers 1.8.0 library, probably beacuse of a fresh installation of the enviroment.  
I use a Mac with Sonoma 14.1.2 but i don't think this is the problem.
I tried to complie and upload with both 1276 and 1262 (even if i have th 1276), but none of the 2 transmitted (of course i have been waiting for a GPS fix etc.).
Do you think it can be realted to the Xpowers library ?

pjalocha, 2024-01-11:
But it is alive, you can see GPS NMEA on the output, etc. ?

WITH_AXP does not worrk ?
I am lost with all these different drivers nd don't have the same T-Beam you have.

tfede37, 2024-01-11:
Update: tried with a different TBEAM board with M8N GPS and at strartub i get the following :
11:34:33.459 -> OGN-Tracker
11:34:33.459 -> I2C scan: 0x34 1 devices
11:34:33.491 -> AXP power/charge chip NOT detected
11:34:33.591 -> RF chip detected
11:34:33.591 -> $POGNS,SysStart*70
11:34:33.591 -> $POGNS,CPU=0x0000B0B21C506594,Address=0x506594,AddrType=3,Stealth=0,AcftType=0x1,FreqPlan=1,TxPower=+14*61
11:34:33.591 -> $POGNS,Pilot=,Crew=,Reg=,Base=*4D
11:34:33.591 -> $POGNS,Manuf=,Model=,Type=,SN=*7E

As far as i see the AXP CHIP cannot be detected or is not recognized.

tfede37, 2024-01-11:
Complete transcription of the boot process:

11:43:13.686 -> E (198) esp_core_dump_flash: No core dump partition found!
11:43:13.686 -> E (199) esp_core_dump_flash: No core dump partition found!
11:43:14.016 -> OGN-Tracker
11:43:14.016 -> I2C scan: 0x34 1 devices
11:43:14.049 -> AXP power/charge chip NOT detected
11:43:14.147 -> RF chip detected
11:43:14.147 -> $POGNS,SysStart*70
11:43:14.147 -> $POGNS,CPU=0x0000B0B21C506594,Address=0x506594,AddrType=3,Stealth=0,AcftType=0x1,FreqPlan=1,TxPower=+14*61
11:43:14.147 -> $POGNS,Pilot=,Crew=,Reg=,Base=*4D
11:43:14.147 -> $POGNS,Manuf=,Model=,Type=,SN=*7E
11:43:14.147 -> $POGNS,Class=,ID=,Task=*3A
11:43:14.147 -> TaskGPS:
11:43:14.147 -> $GNRMC,,V,,,,,,,,,,N*4D

etc. etc...

pjalocha, 2024-01-11:
I find it strange, it does not transmit: when it says it found the RF chip and the GPS has a lock then it should transmit OGN packets. I will check it wiht my hrdware.

tfede37, 2024-01-11:
Thank you Pawel, Just klet me know how i can help

pjalocha, 2024-01-15:
I just revised the code, compiled for T-Beam-S3 and it does transmit OGN status packets.
At startup it prints:
AXP2101 power/charge chip detected
  USB:  5.104V
  Batt: 0.000V
RF chip detected

pjalocha, 2024-01-17:
OK, this explaains why it does not transmit.
Possibly, for this module I need to activate power line for this chip or something like that.

tfede37, 2024-01-18:
In the previous version the RF chip was detected, in the current one it is not. Maybe this helps...

pjalocha, 2024-01-18:
Which exact profile you use ?

tfede37, 2024-01-18:
With the latest release i used the following one:

[env:ttgo-sx1276-tbeam]
board = ttgo-lora32-v1
; board = tbeam
build_flags = -DTTGO_TBEAM
              -DRADIOLIB_GODMODE             ; advanced functions are needed from the RadioLib
              -DWITH_ESP32
              -DWITH_TBEAM10
              -DWITH_SX1276
              -DWITH_CONFIG     ; allow to change parameters via serial console
              -DWITH_GPS_UBX
              -DWITH_GPS_PPS    ; use the PPS of the GPS (not critical but gets betterr timing)
              -DWITH_GPS_CONFIG ; GPS can be adjusted for serial baud rate and navigation model
              -DWITH_AXP
;              -DWITH_BT_SPP     ; BT Standard Serial Port => connection to XCsoar, takes big resources in Flash and RAM
              -DWITH_LOOKOUT
              -DWITH_PFLAA

With the previous one i used the following:
[env:ttgo-sx1276-tbeam]
board = ttgo-lora32-v1
; board = tbeam
build_flags = -DTTGO_TBEAM
              -DRADIOLIB_GODMODE             ; advanced functions are needed from the RadioLib
              -DWITH_ESP32
              -DWITH_TBEAM10
              -DWITH_SX1276
              -DWITH_CONFIG     ; allow to change parameters via serial console
              -DWITH_GPS_UBX
              -DWITH_GPS_PPS    ; use the PPS of the GPS (not critical but gets betterr timing)
              -DWITH_GPS_CONFIG ; GPS can be adjusted for serial baud rate and navigation model
              -DWITH_AXP
;              -DWITH_BT_SPP     ; BT Standard Serial Port => connection to XCsoar, takes big resources in Flash and RAM
              -DWITH_LOOKOUT
              -DWITH_PFLAA

pjalocha, 2024-01-20:
Let's open the case in the prrroper project: ogn-tracker about turrning on power for various elements, as I suspect this is the prroblem here.